natario1/CameraView #132: Fatal Exception: java.lang.RuntimeException startPreview failed

derohimat, 2018-01-05:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. I get this from Crashlytics [here for detail](http://crashes.to/s/04021218714)

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

sahityakumarsuman, 2018-01-16:
derohimat can you tell me how you solved this error

derohimat, 2018-01-16:
@sahityakumarsuman I don't know, because I get this error and now didn't solved

sahityakumarsuman, 2018-01-16:
if anyone solved this one please answer . it will be good help.

natario1, 2018-02-27:
Subscribe or contribute to #122 please

closed 2018-02-27 by natario1