natario1/CameraView #138: Crash on specific device with 'tap to focus'

hiromiuehara, 2018-01-09:
On SM-T813 (samsung tab only for the moment...) once you tap on the screen to focus it launches a 3s delay that calls setParameters to reset the focus. But according to android docs setParameters is not always callable and I've got a runtime exception on the line "setParameters"...

*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Activate tap to focus gesture
2. Tap on screen to focus
3. Wait 3s

#### Expected behavior:

Focus

#### Actual behavior:

Focus and crash after 3s

[image]

vajape, 2018-01-10:
Hi! Casually I was looking for about the same crash exactly. I'm using Samsung Galaxy J5 Android 6.0.1. Usually I use a Nexus 5X Android 8.0.1 wich run perfectly.

hiromiuehara, 2018-02-02:
Do we have some news here @natario1 ? :)

natario1, 2018-02-27:
Can any of you take a look at #167 and test that branch to see if it's fixed? I'll merge into master soon.

natario1, 2018-03-02:
I have released 1.5.0 with that.

natario1, 2018-03-07:
Based on comments I think this is fixed, feel free to reopen.

closed 2018-03-07 by natario1