natario1/CameraView #139: Sometimes wrongly rotated image when saved to File if device orientation != 0

techlung, 2018-01-11:
- [x] *I am using the latest version* (1.4.1 right now)
- [x] Only if modify the demo app a little *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Use certain Tablets or Phones (verified on Samsung SM-T800, Samsung Galaxy J3, Samsung Galaxy Tab A6, Sony Z1) as test devices in Portrait mode. (add android:screenOrientation="portrait" to Activity in Manifest). Bug does not appear on all my devices

2. Save captured picture in onPicture(byte[] jpeg) to File on Device. Save byte Array directly or save Bitmap after using CameraUtils.decodeBitmap(...). E.g:
```
 File file = new File(context.getFilesDir().getAbsolutePath() + "/scans/test.jpg");
OutputStream os = new BufferedOutputStream(new FileOutputStream(file));
bitmap.compress(Bitmap.CompressFormat.JPEG, 100, os);
os.close();
```

3. View Image in PicturePreviewActivty by using File:
`imageView.setImageURI(Uri.fromFile(b)); `

#### Expected behavior:
PicturePreviewActivity showing a normal picture.

#### Actual behavior:
Picture is rotated wrongly sometimes.

This depends on how the devices was tilted while taking the picture. Sometimes the result fits, sometimes it doesn't. When checking the orientation by adding a CameraListener onOrientationChanged(...) it seems like the result always DOESN'T fit if the current device orientation is != 0;

#### Screenshots:
[image]
[image]

natario1, 2018-01-11:
You should do like in the demo app, get the bitmap using CameraUtils (or read the EXIF orientation tag yourself and apply a rotation to the image) and use setImageBitmap

techlung, 2018-01-11:
I need to save the image to the filesystem to store it permanently  and maybe view it again later. I just tested, even if I open the image and read it into a byte array and then pass it to CameraUtils.decodeBitmap and use setImageBitmap it is rotated wrongly. Can the EXIF orientation get lost somewhere on the way? I'm not that familiar with the EXIF tags etc.

natario1, 2018-01-11:
They are included in the byte array. But not in Bitmap objects. You can save the byte array directly to file and then read using CameraUtils

techlung, 2018-01-11:
I believe that's what I'm doing now. I forked it to [`https://github.com/techlung/CameraView`](https://github.com/techlung/CameraView). I can reproduce the effect with exactly that code. Still don't get what I'm doing wrong, sorry :(

natario1, 2018-01-11:
I can't take a look at that now, but if you are saving as we discussed (either the byte array or compressing a Bitmap obtained through CameraUtils) then you just have to show the File. That goes beyond the scope of this library.

Try opening the file with a decent gallery app , it should be looking fine

techlung, 2018-01-11:
I'm now directly writing byte array to public pictures directory and then opening with default Samsung Gallery. Images are still rotated wrongly when I tilt the device to a certain angle. 

I noticed another thing. I get these outputs with different degree values (0,90,180,270) when I tilt the device a little. 
 
com.otaliastudios.cameraview.demo I/CameraCallbacks: onDeviceOrientationChanged 90

I tilt the device just enough to get a Log output but not enough to make the UI actually change orientation. Should these always be in sync, UI and onDeviceOrientationChanged Log outputs? Seems kind of strange to me. Will dig deeper when I get the chance.

natario1, 2018-01-11:
We round degrees, you get 90 when the reported angle from the sensor is between 45 and 135. If you tilt about 45 the result is unpredictable, but this was never a real problem.

If you run out of ideas, try with v1.4.0

techlung, 2018-01-11:
Tried with v.1.4.0. Same issue. I might be mistaken but I believe this issue did not exist with v1.2.1. I need to make use of that fancy FrameProcessor now however so I can't roll back to that. Will check this tomorrow.

natario1, 2018-02-27:
@purposebakery any news on this issue? It's not clear to me what's happening, and I can't reproduce this in the demo app, which is important to figure out a fix.

purposebakery, 2018-02-28:
Sorry, I've been terribly busy. I'll try to reproduce the bug on the demo once I have some free minutes.

purposebakery, 2018-03-18:
With the following changes I can reproduce the issue on the current v2 branch with some slight modifications to the demo app. Forked it [here](https://github.com/purposebakery/CameraView/commit/1b604ad23e83d7c0f3238688031e29d6c142238f) with the according changes.

The issue really only appears on some devices and only if I hold the device horizontally. I attached some photos of my "testing setup" with the Samsung SM-T800. You can see how the CameraActivity shows the image nicely in portrait mode, but the external Gallery app then shows the wrongly rotated image even though both apps are in portrait mode.

I think debugging this one without a "malfunctioning" device will be nearly impossible. I'll try to figure out what other devices don't work.

[image]
[image]

MoustafaElsaghier, 2018-05-14:
Please use this code to get your bitmap

```
CameraUtils.decodeBitmap(byteArray, new CameraUtils.BitmapCallback() {
            @Override
            public void onBitmapReady(Bitmap bitmap) {
                // Do whatever you want with your Bitmap
                // Save it as file or set into ImageView
            }
        });
```
Hope this help you

ridickle7, 2018-08-10:
@MoustafaElsaghier how to captured/saved as captured whether in landscape/portrait mode? 
I want delete "auto orientation detected and auto rotation"...

MoustafaElsaghier, 2018-09-18:
@ridickle7 sorry for late answer. 

but what you mean I misunderstand you, bro.