natario1/CameraView #141: Handling rotation

HD-AD, 2018-01-16:
When rotation the screen the activity gets restarted and it takes some time for the cameraView to be displayed, causing it to show a white screen for 1-2seconds before showing the cameraView again.

I couldn't find a example of how to handle rotation with this library? Maybe I overlooked it?

Any help would be greatly appreciated.

natario1, 2018-01-16:
It's much better if you don't recreate your camera activity for rotations. This is basic Android stuff, just Google it 🙂

The loading time is needed to connect to the hardware, allocate buffers etc.

closed 2018-01-16 by natario1

HD-AD, 2018-01-16:
@natario1 But when I disable screen orientation by adding this to my Activity in the Manifest `android:configChanges="keyboardHidden|orientation|screenSize"` the camera orientation is wrong when I rotate the device.

natario1, 2018-01-16:
With this flag the preview won't change, like every camera app does. Rotating the device will have no effect and that's generally good

natario1, 2018-01-16:
Using button.setRotation() and the orientation callback in the CameraListener

HD-AD, 2018-01-16:
@natario1 `android:configChanges="keyboardHidden|orientation|screenSize"` still doesn't work for me, the screen still rotates, the Activity is not recreated but CameraView still rotates (in the wrong orientation) and if I set `android:screenOrientation="portrait"` then rotation is not picked up in CameraListner callback

natario1, 2018-01-16:
You must use both lines!

HD-AD, 2018-01-16:
Haaa! I got it working, thank you for taking the time to help me and creating a AWSOME library!