natario1/CameraView #142: Unable to get frames while capturing video

ritesh94, 2018-01-16:
While the camera is not capturing photo/video I am able to get the frames using frame processors.
But when I am capturing a video at that time I am unable to get the frames.

Any idea on how to get the frames while capturing the video is in progress.

natario1, 2018-01-16:
If you are not getting them it means that the Camera API does not support this. There's no block from our side as far as I remember.

ritesh94, 2018-01-16:
Below is the code that I am using:

 camera_view.addFrameProcessor(new FrameProcessor() {
                @Override
                public void process(@NonNull Frame frame) {
                      if(mCapturingVideo){
                       System.out.println("Initiated in processMethod "+mCapturingVideo);
                      }
                 }
             });

        camera_view.startRecording();
        mCapturingVideo = true;

// After 15 seconds
      camera_view.stopCapturing();
      camera_view.clearFrameProcessors()
       mCapturingVideo = false;

generally, will I be getting frame while recording is inProgress?

natario1, 2018-01-24:
Looks like this is not supported by the framework. Maybe we can improve this in v2 , #113

closed 2018-01-24 by natario1

bipinvaylu, 2019-01-01:
Hey @ritesh94 Any update on this issue, I am facing same issue on all the devices. @natario1 I have created one demo app to reproduce it. Please find the public repo [here](https://github.com/bipinvayalu/FrameTracker/tree/bipin). When the frame process is working and start video recording. It fails and showing below logs.
Here, I am sharing quick access links:
1) Setting up frame processor - [Click](https://github.com/bipinvayalu/FrameTracker/blob/bipin/app/src/main/java/test/example/frametracker/FrameTracker.java#L25)
2) Start video capturing - [Click](https://github.com/bipinvayalu/FrameTracker/blob/bipin/app/src/main/java/test/example/frametracker/TrackerActivity.java#L70)

Logs:
```
01-01 19:17:05.919 19882-19916/? E/MediaRecorder: start failed: -22
01-01 19:17:05.921 19882-19916/? E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:632)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:874)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:198)
        at android.os.HandlerThread.run(HandlerThread.java:65)
01-01 19:17:05.921 19882-19916/? E/MediaRecorder: stop called in an invalid state: 0
01-01 19:17:06.018 19882-19916/test.example.frametracker E/Camera1: onSurfaceAvailable: Exception while binding camera to preview. java.lang.RuntimeException: setParameters failed
    java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2266)
        at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:153)
        at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:134)
        at com.otaliastudios.cameraview.Camera1.access$500(Camera1.java:26)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:79)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:874)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:198)
        at android.os.HandlerThread.run(HandlerThread.java:65)
01-01 19:17:06.020 19882-19916/test.example.frametracker E/AndroidRuntime: FATAL EXCEPTION: CameraViewController
    Process: test.example.frametracker, PID: 19882
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: setParameters failed
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:82)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:874)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:198)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2266)
        at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:153)
        at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:134)
        at com.otaliastudios.cameraview.Camera1.access$500(Camera1.java:26)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:79)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62) 
        at android.os.Handler.handleCallback(Handler.java:874) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:198) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
01-01 19:17:06.021 19882-19916/test.example.frametracker E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: setParameters failed
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: setParameters failed
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:82)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
        at android.os.Handler.handleCallback(Handler.java:874)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:198)
        at android.os.HandlerThread.run(HandlerThread.java:65)
     Caused by: java.lang.RuntimeException: setParameters failed
        at android.hardware.Camera.native_setParameters(Native Method)
        at android.hardware.Camera.setParameters(Camera.java:2266)
        at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:153)
        at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:134)
        at com.otaliastudios.cameraview.Camera1.access$500(Camera1.java:26)
        at com.otaliastudios.cameraview.Camera1$3.run(Camera1.java:79)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62) 
        at android.os.Handler.handleCallback(Handler.java:874) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:198) 
        at android.os.HandlerThread.run(HandlerThread.java:65) 
```
Please help me to figure it out and let me know if you need anything else.

KevinVegaRem, 2024-12-10:
add this to CameraPreview app:cameraExperimental="true"