natario1/CameraView #144: Geo tagging issue

ankitdubey021, 2018-01-17:
## I'm not getting location details on pic.
this library is very good but i'm facing some issue.
Perhaps you've mentioned that library supports ExifInterface and location tagging on the picture captured by this library. but when i click pic, i'm not getting any info even your sample app is not providing geo location tagging.
please tell me whether it supports or not, if supports then what should i do to get proper location info from pic.
Thanks.

natario1, 2018-01-17:
You must use setLocation, we won't do it for you because we don't want to deal with location permissions etc.

ankitdubey021, 2018-01-17:
i've already used that method.
anyways thank you.

closed 2018-01-24 by natario1