natario1/CameraView #145: Camera capture and preview doesn't match when camera height/width changed from match_parent to anything else.

Ashutosh-Tiwari, 2018-01-17:
I am using the latest version of the library.
The bug (unexpected behaviour) can be reproduced in the Demo App itself.

Steps:
1. Open the demo app, default `width` and `height` of the `camera` would be `match_parent`.
2. Capture a photo of an object with "exact corners".
3. Change the `height` of the `camera` (to any acceptable `height` from available options) from the provided "edit" options
4. Capture the photo of the same object with same "exact corners" you captured last time.

You will see that the first capture was accurate, but the second capture will capture, more than what is visible in the preview, from the top.

Screenshots:
1. match_parent height - preview
[image]

2. match_parent height - capture
[image]

3. 672 height - preview
[image]

4. 672 height - capture
[image]

U can see that the photo captured in 4th screenshot ha more been captured more from top than what is visible in preview, whereas the capture in match_parent params is just fine.

No issues if the capture from bottom is not cropped and captures the whole screen till the bottom. But the top should not capture what is not visible in the preview.
Please help me on this asap. @natario1

natario1, 2018-01-17:
If you want your preview to be equal to the result, use wrap_content.

natario1, 2018-02-27:
I think you just need to set cropOutput to true.

closed 2018-02-27 by natario1