natario1/CameraView #147: Screen Preview Sideways on Nexus 5X and Google Pixel

tsiswandari, 2018-01-17:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Open the App
2. Make sure device is on Auto Rotate setting (instead of portrait)
3. Turn the device clock wise or counter clock wise

#### Expected behavior:
The camera preview should follow the device orientation  preview

#### Actual behavior:
The camera preview on portrait mode is flipped sideways tested on Nexus 5X and Google Pixel (both running android Oreo)

#### Screenshots:
[image]

natario1, 2018-01-17:
Thanks for the report. Could you test with v1.4.0?

tsiswandari, 2018-01-17:
I am having the same issue running on v1.4.0 on both device (Nexus 5X and Google Pixel).
Since this is a fork from Google Camera View, I run the demo app [Google CameraView demo](https://github.com/google/cameraview) and it doesn't seem to have the above orientation issue. I am wondering what could have caused the issue.

natario1, 2018-01-17:
What about 1.3.2? I remember this was working well.

tsiswandari, 2018-01-17:
I just tested with  v1.3.2, 1.3.1 and 1.3.0 tags and they all have the same side ways camera preview on portrait  :(
Works great on other device (SGS7 and LG G6)

natario1, 2018-01-17:
Thanks for helping, I'll take a look when I can.

kevinvanmierlo, 2018-02-13:
@natario1 Probably issue with the camera1 api. The Nexus 5X and presumably the Google Pixel has an inverted camera chip. The camera2 api will handle this correctly, however the camera1 api can't process it properly. I saw that the camera2 api is not yet used right?

natario1, 2018-02-28:
This was an issue with the demo app, because it included `android:configChanges="orientation"` but did not include a `android:orientation`.

You should add both, or none of them.

PR #168 (2018-02-28): Demo app: lock orientation to a fixed value
Fixes #147
closed 2018-02-28 by natario1