natario1/CameraView #148: not able to record video for huawei honor 3c with android version 7.1.2

LtotheWT, 2018-01-22:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. switch to video session
2. click on record video button 
3. not able to record video

#### Expected behavior:
able to record video 

#### Actual behavior:
not able to record video 

#### Screenshots:

#### Interesting logs:
01-22 08:51:43.534 10317-10338/com.otaliastudios.cameraview.demo E/MediaRecorder: start failed: -2147483648
01-22 08:51:43.535 10317-10338/com.otaliastudios.cameraview.demo E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.RuntimeException: start failed.
                                                                            java.lang.RuntimeException: start failed.
                                                                                at android.media.MediaRecorder.start(Native Method)
                                                                                at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:604)
                                                                                at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:58)
                                                                                at android.os.Handler.handleCallback(Handler.java:751)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                at android.os.Looper.loop(Looper.java:154)
                                                                                at android.os.HandlerThread.run(HandlerThread.java:61)
01-22 08:51:43.537 10317-10338/com.otaliastudios.cameraview.demo I/Camera1: endVideoImmediately: is capturing: true
01-22 08:51:43.537 10317-10338/com.otaliastudios.cameraview.demo E/MediaRecorder: stop called in an invalid state: 0
01-22 08:51:43.538 10317-10338/com.otaliastudios.cameraview.demo W/Camera1: endVideoImmediately: Error while closing media recorder. Swallowing java.lang.IllegalStateException
                                                                            java.lang.IllegalStateException
                                                                                at android.media.MediaRecorder.stop(Native Method)
                                                                                at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:634)
                                                                                at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:24)
                                                                                at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:609)
                                                                                at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:58)
                                                                                at android.os.Handler.handleCallback(Handler.java:751)
                                                                                at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                at android.os.Looper.loop(Looper.java:154)
                                                                                at android.os.HandlerThread.run(HandlerThread.java:61)

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

LtotheWT, 2018-01-30:
@natario1 thx for this project, by the way do you have any idea on this ?

natario1, 2018-02-27:
No, no idea, maybe the file you are passing to startRecording is invalid. Which File are you passing?

LtotheWT, 2018-02-28:
hi @natario1, i have testing across many devices, ur library works fine on most device except htc and huawei. End up i use another library that implement camera 2 api.

daniele93dc, 2018-06-15:
i have same error.
`java.lang.IllegalStateException
        at android.media.MediaRecorder._start(Native Method)
        at android.media.MediaRecorder.start(MediaRecorder.java:946)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:638)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:158)
        at android.os.HandlerThread.run(HandlerThread.java:61)`

derohimat, 2018-07-04:
I got same issue run in emulator

natario1, 2018-09-20:
Hoping this was solved by #264 , feel free to reopen.

closed 2018-09-20 by natario1

hangga, 2019-08-19:
I also found the same error on the CUBOT_X18_Plus device

```
**** DEVICE INFORMATION *****
Brand: CUBOT
Device: CUBOT_X18_Plus
Model: CUBOT_X18_Plus
Id: O00623
Product: CUBOT_X18_Plus
*****************************

2019-08-19 14:08:38.626 515-4129/? E/CameraSource: Video dimension (800x480) is unsupported
2019-08-19 14:08:38.628 515-4129/? E/StagefrightRecorder: [setupMPEG4orWEBMRecording] setupMediaSource Fail err=-19
2019-08-19 14:08:38.628 9068-9178/com.sebangsa.hqq.qa E/MediaRecorder: start failed: -19
2019-08-19 14:08:38.631 9068-9178/com.sebangsa.hqq.qa E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:645)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:63)
        at android.os.Handler.handleCallback(Handler.java:869)
        at android.os.Handler.dispatchMessage(Handler.java:101)
        at android.os.Looper.loop(Looper.java:206)
        at android.os.HandlerThread.run(HandlerThread.java:65)
```
Then I do this

```
if (Build.BRAND.toUpperCase().equalsIgnoreCase("CUBOT")){
    cameraPreviewLayout.setVideoQuality(VideoQuality.LOWEST);
} else {
	cameraPreviewLayout.setVideoQuality(VideoQuality.HIGHEST);
}
```

and solved it, Alhamdulillah..