natario1/CameraView #150: When the height of the CameraView is match_parent, the preview is problematic and the image is blurred.

AndroidJohnsonWang, 2018-01-23:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

When the height of the CameraView height is match_parent, the preview is problematic and the image is blurred.

umbrella-alexander-ageychenko, 2018-02-14:
Can approve, when switch to sessionType video preview is stretched (wider than should be)
but scaling is the same as in the picture sessionType, same issue in demo project
Honor 5X, Android 6.0
clear install demo
give permission Direct access to camera image or video capture
open setting bottom sheet
select video session type
give permission to Direct access to the microphone to record audio
everything is fine
open setting bottom sheet - see that picture session type is still selected
select video session type
preview become stretched
(i.e. cup of tee become twice as wide while keeping height and look weird and squished)

itavero, 2018-06-01:
On some Samsung phones I also see this with the quality set to 720p in my own app. They have video resolutions of 4:3 (or 3:4 in portrait mode).