natario1/CameraView #152: Record Video right after change session type from picture to video cause bug (asynchronous tasks)

nguyenvp08, 2018-01-25:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [Yes ] *I am using the latest version*
- [No] *I can reproduce the bug in the demo app, and if not...*
- [ Yes] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:
I see your code with the camera view and the camera controller, all the api you use a handler to post a runnable, especially why you have to restart the camera when set another session type, this action you make 2 inner handler post!!!.

I just try to record the video right after I change the session type to video, but i think sometime the restart camera stop the video recording and bugs happen. 

Can you give me a way that i can record video right after I change from capture image to record video ?
#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

SidhanshuUdawat, 2018-01-29:
@nguyenvp08 what do you mean by bugs ? Are you facing a crash ?

nguyenvp08, 2018-01-29:
hi @SidhanshuUdawat , as I mentioned, when i change the session type from picture to video and do call startRecodingVideo() immediately, I have a problem that sometime the camera will be restart after video is recording and end recording, it's a bug. I know the problem came from the change session type, you also call restart() but the restart() is called in 2 handlers's post, so sometimes restart() is called before startRecording(), sometimes It is called after startRecording()

natario1, 2018-01-29:
The library is asynchronous, if you change session type you must wait for onCameraOpen to be called.