natario1/CameraView #154: CameraView.Start() Should not be on a post runnable

brownpixel, 2018-01-29:
When running the start in a post runnable, we cannot catch any exceptions that may occur when attempting to the start the camera on the device. Exceptions range from Failure to connect, already being in use etc. It's important to handle these exceptions within the app, as it would be better to display a friendly message to the user rather than have the app crash.

- [x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Have another app or your own app be using the camera
2. Open an activity that uses this library and starts the camera in a try catch block
3. Observe the app crashes and cannot catch the exception within the try block

#### Expected behavior:
The app should catch the exception in order to recover and display a friendly message to the user
#### Actual behavior:
App crashes and no exception is thrown to the listener interface

natario1, 2018-01-29:
There is onError() callback, so this is already managed. If the error is not received there, please give us the stacktrace.

brownpixel, 2018-01-29:
The onError() callback does not catch these errors hence the app crashing.

Examples of possible errors:
```
Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:556)
       at android.hardware.Camera.open(Camera.java:372)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:168)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:751)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:154)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

```
Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:156)
       at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:130)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:186)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

natario1, 2018-01-30:
Ok, thanks for the report!

inventionsbyhamid, 2018-03-01:
Getting this exception mostly for OPPO and Xiaomi devices.
``Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:507)
       at android.hardware.Camera.open(Camera.java:343)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:168)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)``

This might be linked to this [Stackoverflow](https://stackoverflow.com/questions/26305107/how-to-fix-fail-to-connect-to-camera-service-exception-in-android-emulator). How to add basic handling and prevent crash?

natario1, 2018-09-14:
This is fixed by #265

closed 2018-09-14 by natario1