natario1/CameraView #155: CameraPreview not showing up.  W/CameraView: onMeasure: surface is not ready. Calling default behavior.

ritesh94, 2018-01-31:
Android Studio 3.1
Micromax HS2 (7.1), Moto G3 (6.0)
Before it was working perfectly. But now below logs keeps coming up. 

```
 W/CameraView: preview: isHardwareAccelerated: true
01-31 19:57:53.405 30379-30379/com.it.wonup.android W/ResourceType: CREATING STRING CACHE OF 1104 bytes
01-31 19:57:53.413 30379-30379/com.it.wonup.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-31 19:57:53.415 30379-30379/com.it.wonup.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-31 19:57:53.436 30379-30379/com.it.wonup.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-31 19:57:53.437 30379-30379/com.it.wonup.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-31 19:57:53.445 30379-30379/com.it.wonup.android W/CameraView: onMeasure: surface is not ready. Calling default behavior.
01-31 19:57:53.449 30379-30379/com.it.wonup.android I/CameraPreview: onSurfaceAvailable: w= 720 h= 720
01-31 19:57:53.450 30379-30379/com.it.wonup.android I/Camera1: onSurfaceAvailable: Size is 720x720
01-31 19:57:54.388 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:57:55.389 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:57:56.128 30379-30379/com.it.wonup.android I/CameraCallbacks: onDeviceOrientationChanged 90
01-31 19:57:56.389 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:57:56.509 30379-30379/com.it.wonup.android I/CameraCallbacks: onDeviceOrientationChanged 0
01-31 19:57:57.390 30379-30461/com.it.wonup.android W/CameraBase: CameraService not published, waiting...
01-31 19:57:57.580 30379-30379/com.it.wonup.android V/Camera1: capturePicture: scheduling
01-31 19:57:57.890 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:57:58.472 30379-30414/com.it.wonup.android V/FA: Inactivity, disconnecting from the service
01-31 19:57:58.890 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:57:59.891 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:00.891 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:01.891 30379-30461/com.it.wonup.android W/CameraBase: CameraService not published, waiting...
01-31 19:58:02.392 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:03.392 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:04.392 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:05.393 30379-30461/com.it.wonup.android I/ServiceManager: Waiting for service media.camera...
01-31 19:58:06.393 30379-30461/com.it.wonup.android W/CameraBase: CameraService not published, waiting...
```

natario1, 2018-01-31:
So what changed between "before" and "now"? We didn't change anything relevant afaik. If it doesn't happen with the demo app, it's likely an application problem.

ritesh94, 2018-01-31:
Nothing changed... Not in the library as well as in my code.... 
Not sure why this happened suddenly!

natario1, 2018-02-27:
I'm closing this, I think it's an app problem. Feel free to open a new one with detailed steps

closed 2018-02-27 by natario1