natario1/CameraView #156: Any listener for when video starts capturing

smoan22, 2018-02-07:
I am changing icon when video recording starts but there is always a 0.5-2 secs gap before video capturing starts which is no issue, but my drawable immediately changes to stop sign which doesn't look good so i needed to know if there was a listener for when video starts capturing so i can change my drawable then and not immediately at the click

natario1, 2018-02-07:
Not yet!

caleb-allen, 2018-02-09:
I am also interested in a feature like this, maybe an `onVideoStarted` in the `CameraListener`.

In `CameraView.startCapturingVideo(File file)` you could pass a callback to `mCameraController.startVideo(file)` which would execute after `mMediaRecorder.start()` completes in `Camera1.java`

I could submit a PR with this if you'd like

caleb-allen, 2018-02-09:
That is, unless I'm mistaken about when the actual recording begins (immediately after `mMediaRecorder.start()`)

natario1, 2018-02-09:
@caleb-allen would be great if you could. You don't need a new callback as you said. You should just add a new method to the CameraView.Callbacks interface. The Camera1 class holds an instance of that, and can call mCallbacks.onVideoRecordingStarted

caleb-allen, 2018-02-09:
Yikes, it looks like all the actual work is hidden under `MediaRecorder.start()` which is a native method, and it appears that it handles work asynchronously as to not block. So I've got that listener in place, but simply executing the callback after calling `mMediaRecorder.start()` isn't giving us an accurate time of when the actual recording starts.

Another idea is to use a [FileObserver](https://developer.android.com/reference/android/os/FileObserver.html) to see (roughly) when data starts streaming into the file.

@natario1 I'm likely going to implement for my use case, although it seems a bit hacky. Let me know if you have any other ideas or if you'd like me to submit a PR with this

natario1, 2018-02-09:
What's the general consensus on this? Like, the Stack Overflow answer about it. How do people get this info?

caleb-allen, 2018-02-09:
There doesn't seem to be much, it seems, my Google-fu may be failing me though. People's solution seems to be just [sleeping until it's recording](https://stackoverflow.com/questions/25649873/android-callback-for-mediarecorder-started-recording)

[This](https://stackoverflow.com/questions/35386663/android-detecting-real-recording-mediarecorder-has-delay) is a pretty close issue, I'm looking into the MediaMetaDataRetriever now.

caleb-allen, 2018-02-09:
[Here](https://stackoverflow.com/questions/25439691/delay-in-preparing-media-recorder-on-android) are a [few more](https://stackoverflow.com/questions/29120902/measuring-android-mediarecorder-delay?rq=1) that don't have any good solutions

caleb-allen, 2018-02-09:
Yep. Been banging my head on this one for a while, Google has yielded no results.

I've tried using a FileObserver and MediaMetaDataRetriever but the problem is that you can't rely on the file changing, because the MediaRecorder holds an in-memory buffer, and it could be any amount of time before it has to flush the buffer to disc. So the only way you could accurately get an idea is if you had access to that buffer, which we don't because it's a native implementation and there are no APIs exposed with that data :(

natario1, 2018-02-09:
I don't see anything suitable for the library itself, but if I had to work on an app, I would try to measure the delay between recorder.start() and actual starting, e.g. by keeping a statistic locally, and improve the estimation over time.

caleb-allen, 2018-02-09:
Yeah. Big shame Google doesn't provide anything for this, my bet is every device model will be just different enough to make it a pain.

Will probably do what you suggest. Cheers

caleb-allen, 2018-02-09:
Alright, last comment. It looks like Camera2 has some ability of better coordinating with the camera with the [CamaeraCaptureSession.StateCallback](https://developer.android.com/reference/android/hardware/camera2/CameraCaptureSession.StateCallback.html)

I may be interested in helping out to implement Camera2 long term to take advantage of things like this.

smoan22, 2018-02-09:
Yeah kinda surprised that there is no listener for this, for now i was just gonna show the icon and start the timer after 1 secs, and then slowly match how long the actual delay is. I can do that by starting the timer and matching it with video's actual time

natario1, 2018-02-27:
I'm closing this, feel free to reopen if something else pops up

closed 2018-02-27 by natario1

androidovshchik, 2018-07-15:
Due to this bug other libraries looks like more interesting