natario1/CameraView #159: Crash

SathishDuo, 2018-02-20:
#### Detailed steps to reproduce the issue:

1.  There Are Three Activities - Activity A - Camera Activity , Activity B - Second Activity , Activity C - Third Activity.
2. Process A to B to C ,while Finishing the Activity C and Starting again Activity A - Camera Activity 
3. When Pressing back button or calling onBackPressd called leads to CRASH

log...

E/CameraController: uncaughtException: Unexpected exception: java.lang.RuntimeException: Fail to connect to camera service
                                                                         java.lang.RuntimeException: Fail to connect to camera service
                                                                             at android.hardware.Camera.<init>(Camera.java:512)
                                                                             at android.hardware.Camera.open(Camera.java:357)
                                                                             at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:168)
                                                                             at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
                                                                             at android.os.Handler.handleCallback(Handler.java:751)
                                                                             at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                             at android.os.Looper.loop(Looper.java:154)
                                                                             at android.os.HandlerThread.run(HandlerThread.java:61)

caleb-allen, 2018-02-20:
Are you overriding `onResume` and `onPause` in your activity? You should be calling `cameraView.start()` in `onResume` and `cameraView.stop()` in `onPause`

natario1, 2018-02-27:
I also think this is an usage issue, feel free to reopen with more info if you have!

closed 2018-02-27 by natario1