natario1/CameraView #164: report a bug - capturePicture() do not take a picture!

yumcoder-platform, 2018-02-25:
- I am using the latest version
- capturePicture() does not take a picture!

Detailed steps to reproduce the issue:
1- call the  camera1.capturePicture()
2- immediately and during capturePicture call the CameraView.toggleFacing()
3- camera1.capturePicture() is disable!

#### Expected behavior:
after call toggleFacing, capturePicture take a picture

#### problem description:
1-   camera1.capturePicture() --> mIsCapturingImage = true;
2-  CameraView.toggleFacing() --> camera1.onStop() --> mCamera.release(); and mIsCapturingImage is never false
3- camera1.capturePicture() does not work beacuse when mIsCapturingImage==true capturePicture return immediately

#### how to fix it:
in the last line of camera1.onStop() add the following line:
 **mIsCapturingImage = false;**

natario1, 2018-02-27:
I don't think `mIsCapturingImage` has a role here, the operations are executed in their order.

But maybe this can be improved.

natario1, 2018-02-27:
For the moment, you should wait for the picture callback before using toggleFacing.

natario1, 2018-03-01:
Thanks, I understand what you were saying now that I took a look. It should be fixed as you said! #173

PR #173 (2018-03-01): Reset flags when stopping camera
Fixes #164
closed 2018-03-01 by natario1