natario1/CameraView #169: How to set crop width and height?

asadullahrao, 2018-02-28:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.
2.
3.

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

guness, 2018-03-01:
I guess you can just set width and height of the preview and let the library crop output for you.

xsacha, 2018-03-07:
Is there a way to prevent it from doing a center crop?
I would like to view exactly the top of the image as that's the section I process in my jni library.

natario1, 2018-03-20:
@xsacha no way, sorry, you can only use `wrap_content` to do no crop at all. We crop the preview with a centerCrop behavior.

closed 2018-03-20 by natario1