natario1/CameraView #171: When orientation is locked by XML, camera orientation is still able to change

guness, 2018-03-01:
- [x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

This is rather a question than a bug.
Whan I lock the orientation of the camera activity by
```
<activity
            android:name=".ui.record.RecordActivity"
            android:configChanges="orientation|keyboardHidden|screenSize"
            android:screenOrientation="landscape"
            android:theme="@style/FullscreenTheme" />
```
I would expect camera orientation is also locked, however when I start recording video portrait, the output is also portrait and vice versa regardless of the activity orientation. I was not able to find a method to set(or lock) orientation manually as well. 

I like this smart feature, however requirement on my current work forces me to have a locked video orientation. So I am humbly asking if this is possible?

#### Screenshots:

[image]

natario1, 2018-03-01:
There is no such method, if you start recording holding the phone in landscape, we will respect that orientation (no matter the activity orientation).

If we didn't do that, your video would appear rotated (for example, sky in the left/bottom part instead of top). If you wish, you can rotate the video afterwards using some other tool.

guness, 2018-03-02:
Got it thanks!

closed 2018-03-02 by guness