natario1/CameraView #176: integrating with CyberAgent/android-gpuimage

yumcoder-platform, 2018-03-05:
How to integrating CameraView with [CyberAgent/android-gpuimage](https://github.com/CyberAgent/android-gpuimage)?

natario1, 2018-03-05:
I am working at it. This would be super cool.

natario1, 2019-08-06:
We now have real-time filters thanks to #527 .

closed 2019-08-06 by natario1

sidhuparas, 2020-07-22:
@natario1 Thanks for this awesome library. It works flawlessly, kudos! I have a query. I tried the real-time filters which are nice but I would like to use some filters from `gpu-image`, for example this:

```
public class GlToneFilter extends GlThreex3TextureSamplingFilter {

    public static final String FRAGMENT_SHADER =
            "precision highp float;\n" +

                    "uniform lowp sampler2D sTexture;\n" +

                    "varying vec2 textureCoordinate;\n" +
                    "varying vec2 leftTextureCoordinate;\n" +
                    "varying vec2 rightTextureCoordinate;\n" +

                    "varying vec2 topTextureCoordinate;\n" +
                    "varying vec2 topLeftTextureCoordinate;\n" +
                    "varying vec2 topRightTextureCoordinate;\n" +

                    "varying vec2 bottomTextureCoordinate;\n" +
                    "varying vec2 bottomLeftTextureCoordinate;\n" +
                    "varying vec2 bottomRightTextureCoordinate;\n" +

//			"uniform highp float intensity;" +
                    "uniform highp float threshold;" +
                    "uniform highp float quantizationLevels;" +

                    "const highp vec3 W = vec3(0.2125, 0.7154, 0.0721);" +

                    "void main() {\n" +
                    "vec4 textureColor = texture2D(sTexture, textureCoordinate);" +

                    "float bottomLeftIntensity = texture2D(sTexture, bottomLeftTextureCoordinate).r;" +
                    "float topRightIntensity = texture2D(sTexture, topRightTextureCoordinate).r;" +
                    "float topLeftIntensity = texture2D(sTexture, topLeftTextureCoordinate).r;" +
                    "float bottomRightIntensity = texture2D(sTexture, bottomRightTextureCoordinate).r;" +
                    "float leftIntensity = texture2D(sTexture, leftTextureCoordinate).r;" +
                    "float rightIntensity = texture2D(sTexture, rightTextureCoordinate).r;" +
                    "float bottomIntensity = texture2D(sTexture, bottomTextureCoordinate).r;" +
                    "float topIntensity = texture2D(sTexture, topTextureCoordinate).r;" +
                    "float h = -topLeftIntensity - 2.0 * topIntensity - topRightIntensity + bottomLeftIntensity + 2.0 * bottomIntensity + bottomRightIntensity;" +
                    "float v = -bottomLeftIntensity - 2.0 * leftIntensity - topLeftIntensity + bottomRightIntensity + 2.0 * rightIntensity + topRightIntensity;" +

                    "float mag = length(vec2(h, v));" +
                    "vec3 posterizedImageColor = floor((textureColor.rgb * quantizationLevels) + 0.5) / quantizationLevels;" +
                    "float thresholdTest = 1.0 - step(threshold, mag);" +
                    "gl_FragColor = vec4(posterizedImageColor * thresholdTest, textureColor.a);" +
                    "}";

    private float threshold = 0.2f;
    private float quantizationLevels = 10f;

    public GlToneFilter() {
        super(FRAGMENT_SHADER);
    }

    //////////////////////////////////////////////////////////////////////////

    public float getThreshold() {
        return threshold;
    }

    public void setThreshold(final float threshold) {
        this.threshold = threshold;
    }

    public float getQuantizationLevels() {
        return quantizationLevels;
    }

    public void setQuantizationLevels(final float quantizationLevels) {
        this.quantizationLevels = quantizationLevels;
    }

    //////////////////////////////////////////////////////////////////////////

    @Override
    public void onDraw() {
        GLES20.glUniform1f(getHandle("threshold"), threshold);
        GLES20.glUniform1f(getHandle("quantizationLevels"), quantizationLevels);
    }
}
```

This didn't work with CameraView filter obviously so I extended from BaseFilter and made some changes (don't know much about OpenGL so probably made temporary non-working changes just to make the code run-able for eg. returning some float value in an overridden function), but it keeps throwing error that `textureCoordinate` isn't set. It seems that such filters are not supported. Can you help me out?

natario1, 2020-07-22:
I don't know what's happening, overall what you are doing should be possible but it's not trivial - it could take you a few days of trial and error and going through source code, given that you are not familiar with OpenGL.

sidhuparas, 2020-07-22:
@natario1 Thanks for the quick reply. Yes it seems to be the case with me. For the workaround I tried `GPUCameraRecorder` class which worked but I missed CameraView's features. I am still looking into it. If you get some idea or pointer, please post it then :)