natario1/CameraView #178: How to set biggest preview size?

HanderWei, 2018-03-07:
I need to capture the biggest size of Picture from preview. But I found that I can't set preview size outside.

natario1, 2018-03-07:
Why would you do this? Can't you use `setPictureSize` and choose a reasonable size there?

HanderWei, 2018-03-08:
@natario1 I use it for image processing. But the default preview size can't match my requirement.

I found that googlesamples did the same thing.

[https://github.com/googlesamples/android-Camera2Basic/blob/master/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.java#L511-L518](https://github.com/googlesamples/android-Camera2Basic/blob/master/Application/src/main/java/com/example/android/camera2basic/Camera2BasicFragment.java#L511-L518)

```
// For still image captures, we use the largest available size.
Size largest = Collections.max(
Arrays.asList(map.getOutputSizes(ImageFormat.JPEG)),
new CompareSizesByArea());
mImageReader = ImageReader.newInstance(largest.getWidth(), largest.getHeight(),
ImageFormat.JPEG, /*maxImages*/2);
mImageReader.setOnImageAvailableListener(
mOnImageAvailableListener, mBackgroundHandler);
```

natario1, 2018-03-08:
That is the JPEG output size, which in this library you can control with `setPictureSize()` or `cameraPictureSize*` XML attributes. The default is `SizeSelectors.biggest()` so we are already there.

The preview size is automatically set as something that respect the output aspect ratio and is big enough to be shown in your `View`, because anything bigger than your `View` is a waste of memory resources.

There is an open PR #134 which does what you do, you can use that, I don't think we are going to merge it, it would mess up with measuring behavior with little benefit to just a few users.

closed 2018-03-20 by natario1