natario1/CameraView #179: Generating bitmap using byte array from FrameProcessor is always null

inventionsbyhamid, 2018-03-09:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Use FrameProcessor in CameraView 
2. Decode the byte array to bitmap
3. bitmap is always null

#### Expected behavior: bitmap should not be null ?

#### Actual behavior: bitmap is always null

#### Screenshots:

#### Interesting logs:

The Bitmap after decoding is always null using the frame.getData() byte array. I tried using both BitmapFactory.decodeByteArray and CameraUtils.decodeBitmap.

natario1, 2018-03-09:
This is not possible, the preview frame is not a JPEG image, it is in `ImageFormat.NV21` format (frame.getFormat()). It's not supported by `BitmapFactory`. You can use the android `YuvImage` class to convert but that's really expensive to do for every frame.

inventionsbyhamid, 2018-03-09:
How do I achieve this? I want to create bitmaps for some frames. Decoding to Bitmap gives blurred image (with green and pink colors). Do i need to modify byte array before creating bitmap? I tried using this https://stackoverflow.com/questions/35826709/yuv-nv21-image-converting-to-bitmap

natario1, 2018-03-09:
The byte array is what we get from the camera, without modifying. There must be something wrong in your process, your best bet is to ask on stackoverflow indeed

inventionsbyhamid, 2018-03-11:
Thanks for your help. I was able to figure out the issue. The height width I was using to convert the byte array to bitmap was different than frame height width. I was setting camera view height width to be equal to device width and used the same value in conversion which happened to be slightly different in frame height width.

closed 2018-03-11 by inventionsbyhamid

SahSantoshh, 2018-12-25:
> _If you are trying to report a bug, please file the information below. Prerequisites:_
>
>     * [x]  _I am using the latest version_
>
>     * [ ]  _I can reproduce the bug in the demo app, and if [...]

its same here

SahSantoshh, 2018-12-25:
> Thanks for your help. I was able to figure out the issue. The height width I was using to convert the byte array to bitmap was different than frame height width. I was setting camera view height [...]

can you share your code how you fixed it.