natario1/CameraView #181: Bug in tablet Galaxy Tab A (2016) 7.0 Wi-Fi Focus

RonaldSalazarH, 2018-03-14:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Install the page's apk on a Tablet (Galaxy Tab A (2016) 7.0 Wi-Fi)
2. I took a picture with focus (focusWithMarker)
3. The application remains frozen, it does not advance to the preview of the photo taken
4. In the case of cell phones, it works correctly.

#### Expected behavior:

The application remains frozen, it does not advance to the preview of the photo taken

natario1, 2018-03-20:
Thanks for the report!

mahdi-ninja, 2018-11-19:
I have the same exact issue on Galaxy Tab A (SM-T280).

* Device info
```
Device Model: SM-T280
Architecture: armv7l
OS Build: T280XXU0AQD1
OS Version: 5.1.1
```

- Stack trace:
```
java.lang.RuntimeException: autoFocus failed
	at android.hardware.Camera.native_autoFocus()(Camera.java:-2)
	at android.hardware.Camera.autoFocus()(Camera.java:1331)
	at com.otaliastudios.cameraview.Camera1$19.run()(Camera1.java:826)
	at com.otaliastudios.cameraview.Camera1$2.run()(Camera1.java:63)
	at android.os.Handler.handleCallback()(Handler.java:739)
	at android.os.Handler.dispatchMessage()(Handler.java:95)
	at android.os.Looper.loop()(Looper.java:145)
	at android.os.HandlerThread.run()(HandlerThread.java:61)
```

I don't think there's anyway for us to handle this properly. Can we just simple wrap the [call to `mCamera.autoFocus`](https://github.com/natario1/CameraView/blob/d0bb8cd31cf4fe263be8c95dec314fbdf8a9994c/cameraview/src/main/java/com/otaliastudios/cameraview/Camera1.java#L826) in a try/catch block and ignore the exception?

mahdi-ninja, 2018-11-19:
Looks like this is not specific to tablets. [Here](https://github.com/dm77/barcodescanner/issues/7) people were reporting the same issue on Sony phones as well and [here's](https://github.com/dm77/barcodescanner/blob/ffeaeaf8a0fc2ff247056eb731cb0500df7c2dc4/core/src/main/java/me/dm7/barcodescanner/core/CameraPreview.java#L105) their solution.
```
    public void safeAutoFocus() {
        try {
            mCameraWrapper.mCamera.autoFocus(autoFocusCB);
        } catch (RuntimeException re) {
            // Horrible hack to deal with autofocus errors on Sony devices
            // See https://github.com/dm77/barcodescanner/issues/7 for example
            scheduleAutoFocus(); // wait 1 sec and then do check again
        }
    }

    private void scheduleAutoFocus() {
        mAutoFocusHandler.postDelayed(doAutoFocus, 1000);
    }
```

Although to me, it seems like an overkill and just ignoring the error should be enough.

wosiu, 2018-12-05:
+1 for scheduling one more time

natario1, 2018-12-16:
So we're treating this as a device specific issue - CameraView now swallows this exception and calls the auto focus callback with success=false. At that point you can schedule another focus pass if you want.

closed 2018-12-16 by natario1