natario1/CameraView #183: Multiples errors Camera1 and CameraController

Rodrigolmti, 2018-03-16:
Hello I using you lib for catch some photos in my app and there are multiples users getting errors:

Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.<init>(Camera.java:528)
       at android.hardware.Camera.open(Camera.java:369)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:168)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:156)
       at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:130)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:186)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:175)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)

Some of devices getting this error:
Galaxy J550
Galaxy Core233
Galaxy J5(2016)
LGE4
TCL3
motorola
samsung

[Unknown.txt](https://github.com/natario1/CameraView/files/1820850/Unknown.txt)
[Unknown-2.txt](https://github.com/natario1/CameraView/files/1820852/Unknown-2.txt)

Could you please help me?

umbrella-alexander-ageychenko, 2018-03-30:
also have some crashes on Samsung Galaxy S4 (jfltevzw), 2048MB RAM, Android 4.4

java.lang.RuntimeException: 
  at android.hardware.Camera.startPreview (Native Method)
  at com.jellychip.view.customView.cameraView.java.com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview (Camera1.java:174)
  at com.jellychip.view.customView.cameraView.java.com.otaliastudios.cameraview.Camera1.bindToSurface (Camera1.java:148)
  at com.jellychip.view.customView.cameraView.java.com.otaliastudios.cameraview.Camera1.onStart (Camera1.java:211)
  at com.jellychip.view.customView.cameraView.java.com.otaliastudios.cameraview.CameraController$3.run (CameraController.java:189)
  at android.os.Handler.handleCallback (Handler.java:733)
  at android.os.Handler.dispatchMessage (Handler.java:95)
  at android.os.Looper.loop (Looper.java:157)
  at android.os.HandlerThread.run (HandlerThread.java:61)

natario1, 2018-04-07:
Please subscribe to #189 and #122 for the two issues, so we have separate tickets.

Also, the best thing to identify what's wrong would be to use `CameraLogger` and send log messages to your server, if any of you can do that, it would be great.

closed 2018-04-07 by natario1