natario1/CameraView #184: Camera got stucked in Oneplus devices.

saumilpetechmavens, 2018-03-17:
-I am using the latest version.

Here are the steps to reproduce the issue:
1.capture some pictures.
2.After close the application and come back again.
3.Than Demo app starts the camera for 1 second and than it will stuck and last preview will show in the screen here I am attaching GIF of the issue that show the screen got stuck while device is moving but it is not display any thing.

Note: this issue are only come in One+ device.
Device Detail:
Device : oneplus 3
Android OS : 8.0
[image]

natario1, 2018-04-07:
Can you post cameraview logs from logcat while this occurs?

saumilpetechmavens, 2018-04-09:
I am getting this line multiple times in Log.

E/mm-camera: <IFACE ><ERROR> 2129: iface_axi_fill_framedrop_struct_from_bufq_handle: Error! Uncleared Error mask 0x20 frame_id 88.

kunaldawn, 2018-05-09:
Hi @saumilpetechmavens this issue is reproduced for me consistently. The issue is with back camera only. Once the issue is triggered from your demo app, it affects system default camera app also. The issue goes away only after phone is rebooted.

Switching to front camera causes this error, view is frozen 
```
05-09 12:03:56.279 4935-24365/? E/mm-camera: <PPROC ><ERROR> 689: base_port_handle_buf_divert_ack: base_port_handle_buf_divert_ack:689, invalid condition, no ack found in list for identity=20002, frameid=3
05-09 12:03:56.279 4935-24365/? E/mm-camera: <IMGLIB><ERROR> 565: module_imgbase_forward_port_event: module_imgbase_forward_port_event_paaf:565] Fowarding event 2 from src port failed
05-09 12:03:56.279 4935-24365/? E/mm-camera: <CPP   ><ERROR> 890: cpp_module_send_event_upstream: send_event_to_peer failed 0
05-09 12:03:56.279 4935-24365/? E/mm-camera: <CPP   ><ERROR> 966: cpp_module_send_buf_divert_ack: [CPP_BUF:] send_event_upstream failed -14
```

After that following error is thrown continiously
```
05-09 12:06:38.616 4935-24400/? E/mm-camera: <IFACE ><ERROR> 2129: iface_axi_fill_framedrop_struct_from_bufq_handle: Error! Uncleared Error mask 0x20 frame_id 2708
```

Device Information
Model : One Plus 3T
Variant : A3003
Android Version : 8.0.0
OxygenOS Version : 5.0.2

### Plz give some light on this issue. i am stuck with on my production build. Your help will be very much appreciated.

natario1, 2018-10-10:
@pandulapeter check the logcat in the demo app and try see what's wrong there

pandulapeter, 2018-10-10:
I can see the same logs over and over again that @kunaldawn posted. Same device with same Android version and the issue only appears when using "video" session type - "picture" works fine on the OnePlus 3T and both session types work fine on other devices I tested on.

natario1, 2018-10-10:
Thanks. So it's not something I understand / can fix. See if StackOverflow helps about it

pandulapeter, 2018-10-10:
Ok, thanks for your reply! 👍

zhuzhao404, 2018-11-28:
> I can see the same logs over and over again that @kunaldawn posted. Same device with same Android version and the issue only appears when using "video" session type - "picture" works fine on the OnePlus 3T and both session types work fine on other devices I tested on.

Hi,friend!Did you slove this?Thanks

zhuzhao404, 2018-11-28:
> > Hi,friend!Did you slove this?Thanks
> 
> Sorry, the only news that I have is that there was one device where a factory reset solved the issue.

All right，thanks for your reply

natario1, 2018-12-16:
> Sorry, the only news that I have is that there was one device where a factory reset solved the issue.

I would say this is out of our control. Closing for now.

closed 2018-12-16 by natario1