natario1/CameraView #189: Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service

chiyan171, 2018-03-30:
Version：compile 'com.otaliastudios:cameraview:1.5.0'
OPPO A57 Android 6.0
Fatal Exception: java.lang.RuntimeException: Fail to connect to camera service
       at android.hardware.Camera.(Camera.java:570)
       at android.hardware.Camera.open(Camera.java:384)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:172)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:180)
       at android.os.Handler.handleCallback(Handler.java:739)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:179)
       at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2018-04-07:
Thanks for the report. Any more info to share?
- steps to reproduce
- when does this happen
- logs. Looks like you can reproduce this, so please add cameraview logs from logcat

enister, 2018-04-27:
I have the same error on Meizu MX6 (Android 6.0) when i want to open the front camera in my activity.
All permissions are granted.

```
Fatal Exception: java.lang.RuntimeException
Fail to connect to camera service
android.hardware.Camera.<init> (Camera.java:669)
android.hardware.Camera.open (Camera.java:500)
com.otaliastudios.cameraview.Camera1.onStart (Camera1.java:170)
com.otaliastudios.cameraview.CameraController$3.run (CameraController.java:178)
android.os.Handler.handleCallback (Handler.java:815)
android.os.Handler.dispatchMessage (Handler.java:104)
android.os.Looper.loop (Looper.java:207)
android.os.HandlerThread.run (HandlerThread.java:61)
```

MoustafaElsaghier, 2018-05-14:
@natario1  have you solved it?
it appears on nexus-4 5.1.1 too

HBiSoft, 2018-06-19:
Same issue on Motorola Moto X (1st Gen) (ghost), 2048MB RAM, Android 5.1

```
java.lang.RuntimeException: 
  at android.hardware.Camera.<init> (Camera.java:528)
  at android.hardware.Camera.open (Camera.java:369)
  at com.otaliastudios.cameraview.c.c (Camera1.java:168)
  at com.otaliastudios.cameraview.d$3.run (CameraController.java:175)
  at android.os.Handler.handleCallback (Handler.java:739)
  at android.os.Handler.dispatchMessage (Handler.java:95)
  at android.os.Looper.loop (Looper.java:135)
  at android.os.HandlerThread.run (HandlerThread.java:61)
```

Any feedback on this issue, my app is currently live and would like to resolve this.

natario1, 2018-06-19:
See https://github.com/natario1/CameraView/issues/189#issuecomment-379462951

HBiSoft, 2018-06-19:
@chiyan171 @enister @MoustafaElsaghier Can one of you please give more details so the issue can be resolved? I can't reproduce the issue on my device.

HBiSoft, 2018-06-19:
@natario1 The issue is not related to front camera because the device that crashed has a front facing camera, but just to make sure I added

```
if(getApplicationContext().getPackageManager().hasSystemFeature(PackageManager.FEATURE_CAMERA_FRONT)) {
    cameraView.setFacing(Facing.FRONT);
}
```

I will keep trying to reproduce the issue and let you know if I was successful or not.

HBiSoft, 2018-07-12:
Did anyone find the issue? I'm still getting crashes and will have to look at another library if I can't resolve this soon.

I tried reproducing the crash, but failed to do so.

KORuL, 2018-08-05:
Same issue, and i can get logs from CameraLogger.
[image]

KORuL, 2018-08-05:
[image]

Aimank96, 2018-08-05:
same issue here,on samsung j1 ace

HBiSoft, 2018-08-05:
@KORuL thank you for the log, can you please post the full log from logcat? It's the only way the owner of this library can fix the issue.

KORuL, 2018-08-05:
2018-08-05 14:56:19.235 1387-1412/? W/audio_hw_generic: Not supplying enough data to HAL, expected position 12739738 , only wrote 12739680
2018-08-05 14:56:19.549 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.InterView.showInterView(android.content.Context, java.lang.String, java.lang.String) took 305.394ms
    
    --------- beginning of system
2018-08-05 14:56:19.621 1714-2140/system_process I/ActivityManager: START u0 {flg=0x10000000 cmp=com.some.packet/com.blocknotary.interview.ActivityMain (has extras)} from uid 10087 on display 0
2018-08-05 14:56:19.631 17530-17535/com.some.packet I/art: Do full code cache collection, code=251KB, data=149KB
2018-08-05 14:56:19.632 17530-17535/com.some.packet I/art: Starting a blocking GC JitCodeCache
2018-08-05 14:56:19.632 17530-17535/com.some.packet I/art: After code cache collection, code=126KB, data=84KB
2018-08-05 14:56:19.661 1714-1842/system_process I/ActivityManager: START u0 {flg=0x10000000 cmp=com.some.packet/com.blocknotary.interview.ActivityMain (has extras)} from uid 10087 on display 0
2018-08-05 14:56:20.001 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.closeLoading(boolean) took 227.131ms
2018-08-05 14:56:20.164 1714-1732/system_process W/ActivityManager: Activity pause timeout for ActivityRecord{d0f1265 u0 com.some.packet/com.blocknotary.interview.ActivityMain t95}
2018-08-05 14:56:20.259 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.handleShutter() took 111.822ms
2018-08-05 14:56:20.434 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.initButtonsListener() took 174.889ms
2018-08-05 14:56:20.631 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.initText(boolean) took 195.336ms
2018-08-05 14:56:20.719 17530-17537/com.some.packet I/art: Starting a blocking GC Instrumentation
2018-08-05 14:56:20.815 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.initializeCamera(com.blocknotary.interview.ActivityMain) took 183.741ms
2018-08-05 14:56:21.374 17530-17530/com.some.packet W/art: Verification of void com.blocknotary.interview.ActivityMain.onResume() took 270.894ms
2018-08-05 14:56:21.709 17530-17541/com.some.packet I/art: Background sticky concurrent mark sweep GC freed 33255(2MB) AllocSpace objects, 4(80KB) LOS objects, 7% free, 15MB/16MB, paused 8.299ms total 92.311ms
2018-08-05 14:56:21.800 1714-1728/system_process I/GnssLocationProvider: WakeLock acquired by sendMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@8d87df4)
2018-08-05 14:56:21.814 17530-17530/com.some.packet D/FFmpeg: Build.CPU_ABI : x86
2018-08-05 14:56:21.814 17530-17530/com.some.packet I/FFmpeg: Loading FFmpeg for x86 CPU
2018-08-05 14:56:21.824 1388-1388/? V/EmulatedCamera_Camera: getCameraInfo
2018-08-05 14:56:21.825 1388-1388/? V/EmulatedCamera_BaseCamera: getCameraInfo
2018-08-05 14:56:21.832 1388-2370/? I/CameraService: CameraService::connect call (PID -1 "com.some.packet", camera ID 0) for HAL version default and Camera API version 1
2018-08-05 14:56:21.979 1714-1731/system_process I/GnssLocationProvider: WakeLock released by handleMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@8d87df4)
2018-08-05 14:56:22.010 17530-17643/com.some.packet W/art: Verification of void com.github.hiteshsondhi88.libffmpeg.FileUtils.<init>() took 116.158ms
2018-08-05 14:56:22.034 17530-17541/com.some.packet I/art: Background partial concurrent mark sweep GC freed 7927(479KB) AllocSpace objects, 4(80KB) LOS objects, 16% free, 19MB/23MB, paused 5.293ms total 165.316ms
2018-08-05 14:56:22.040 17530-17535/com.some.packet I/art: Do partial code cache collection, code=209KB, data=164KB
2018-08-05 14:56:22.045 17530-17535/com.some.packet I/art: After code cache collection, code=206KB, data=162KB
2018-08-05 14:56:22.046 17530-17535/com.some.packet I/art: Increasing code cache capacity to 1024KB
2018-08-05 14:56:22.046 17530-17535/com.some.packet I/art: JIT allocated 60KB for compiled code of void android.widget.TextView.<init>(android.content.Context, android.util.AttributeSet, int, int)
2018-08-05 14:56:22.477 17530-17530/com.some.packet I/Choreographer: Skipped 168 frames!  The application may be doing too much work on its main thread.
2018-08-05 14:56:22.635 1714-3391/system_process I/GnssLocationProvider: WakeLock acquired by sendMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@ef98560)
2018-08-05 14:56:22.637 17530-17530/com.some.packet D/FFmpeg: Build.CPU_ABI : x86
2018-08-05 14:56:22.637 17530-17530/com.some.packet I/FFmpeg: Loading FFmpeg for x86 CPU
2018-08-05 14:56:22.643 1714-1731/system_process I/GnssLocationProvider: WakeLock released by handleMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@ef98560)
2018-08-05 14:56:22.726 17530-17622/com.some.packet D/EGL_emulation: eglMakeCurrent: 0xae3070a0: ver 3 0 (tinfo 0xae303920)
2018-08-05 14:56:22.852 1387-1413/? W/audio_hw_generic: Not supplying enough data to HAL, expected position 13084838 , only wrote 12913200
2018-08-05 14:56:22.897 17530-17622/com.some.packet D/EGL_emulation: eglMakeCurrent: 0xae3070a0: ver 3 0 (tinfo 0xae303920)
2018-08-05 14:56:22.915 17530-17622/com.some.packet D/EGL_emulation: eglMakeCurrent: 0xae3070a0: ver 3 0 (tinfo 0xae303920)
[... cut ...]
2018-08-05 14:56:30.951 17530-17616/com.some.packet I/CrashlyticsCore: Crashlytics report upload complete: 5B66E5D702F4-0001-447A-02DD1808E2A2
2018-08-05 14:56:34.630 2599-17547/com.google.android.googlequicksearchbox:search W/GmsLocationProvider: Error removing location updates: 16
2018-08-05 14:56:40.163 1714-1732/system_process W/ActivityManager: Launch timeout has expired, giving up wake lock!
2018-08-05 14:57:04.976 2599-2599/com.google.android.googlequicksearchbox:search I/GrammarCompilationSvcCt: #startService for null, APP_NAMES.

KORuL, 2018-08-05:
it is a vm NEXUS 4 API 25

HBiSoft, 2018-08-05:
@natario1 please look at the log above.

KORuL, 2018-08-05:
This VM was worked fine, but i change orientation and this vm start give me this exception

KORuL, 2018-08-05:
More helpfull

2018-08-05 15:34:42.366 1387-1412/? W/audio_hw_generic: Not supplying enough data to HAL, expected position 12913259 , only wrote 12913200
2018-08-05 15:34:42.478 18137-18144/com.some.packet I/art: Starting a blocking GC Instrumentation
2018-08-05 15:34:42.482 18137-18137/com.some.packet W/art: Verification of void com.some.interview.InterView.showInterView(android.content.Context, java.lang.String, java.lang.String) took 106.140ms
    
    --------- beginning of system
2018-08-05 15:34:42.563 1714-1842/system_process I/ActivityManager: START u0 {flg=0x10000000 cmp=com.some.packet/com.some.interview.ActivityMain (has extras)} from uid 10087 on display 0
2018-08-05 15:34:43.391 18137-18137/com.some.packet W/art: Verification of void com.some.interview.ActivityMain.onResume() took 215.857ms
2018-08-05 15:34:43.649 1714-1728/system_process I/GnssLocationProvider: WakeLock acquired by sendMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@3bb1996)
2018-08-05 15:34:43.655 18137-18137/com.some.packet D/FFmpeg: Build.CPU_ABI : x86
2018-08-05 15:34:43.655 18137-18137/com.some.packet I/FFmpeg: Loading FFmpeg for x86 CPU
2018-08-05 15:34:43.656 18137-18137/com.some.packet W/WorkerHandler: get: Reusing cached worker handler. CameraViewController
2018-08-05 15:34:43.657 18137-18137/com.some.packet V/WorkerHandler: get: Reusing cached worker handler. CameraViewController
2018-08-05 15:34:43.657 18137-18137/com.some.packet W/WorkerHandler: get: Reusing cached worker handler. CameraViewWorker
2018-08-05 15:34:43.657 18137-18137/com.some.packet V/WorkerHandler: get: Reusing cached worker handler. CameraViewWorker
2018-08-05 15:34:43.658 18137-18137/com.some.packet W/WorkerHandler: get: Reusing cached worker handler. FrameProcessorsWorker
2018-08-05 15:34:43.658 18137-18137/com.some.packet V/WorkerHandler: get: Reusing cached worker handler. FrameProcessorsWorker
2018-08-05 15:34:43.664 1714-1731/system_process I/GnssLocationProvider: WakeLock released by handleMessage(3, 0, com.android.server.location.GnssLocationProvider$GpsRequest@3bb1996)
2018-08-05 15:34:43.673 18137-18137/com.some.packet I/CameraController: Start: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.673 18137-18137/com.some.packet V/CameraController: Start: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.673 18137-18137/com.some.packet V/CameraController: Start: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.674 18137-18137/com.some.packet I/CameraController: Stop: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.674 18137-18137/com.some.packet V/CameraController: Stop: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.674 18137-18137/com.some.packet V/CameraController: Stop: posting runnable. State: STATE_STOPPED
2018-08-05 15:34:43.675 18137-18192/com.some.packet I/CameraController: Start: executing. State: STATE_STOPPED
2018-08-05 15:34:43.675 18137-18192/com.some.packet V/CameraController: Start: executing. State: STATE_STOPPED
2018-08-05 15:34:43.675 18137-18192/com.some.packet V/CameraController: Start: executing. State: STATE_STOPPED
2018-08-05 15:34:43.675 18137-18192/com.some.packet I/CameraController: Start: about to call onStart() STATE_STARTING
2018-08-05 15:34:43.675 18137-18192/com.some.packet V/CameraController: Start: about to call onStart() STATE_STARTING
2018-08-05 15:34:43.675 18137-18192/com.some.packet V/CameraController: Start: about to call onStart() STATE_STARTING
2018-08-05 15:34:43.675 1388-1388/? V/EmulatedCamera_Camera: getCameraInfo
2018-08-05 15:34:43.675 1388-1388/? V/EmulatedCamera_BaseCamera: getCameraInfo
2018-08-05 15:34:43.676 18137-18137/com.some.packet I/CameraController: Start: posting runnable. State: STATE_STARTING
2018-08-05 15:34:43.676 18137-18137/com.some.packet V/CameraController: Start: posting runnable. State: STATE_STARTING
2018-08-05 15:34:43.676 18137-18137/com.some.packet V/CameraController: Start: posting runnable. State: STATE_STARTING
2018-08-05 15:34:43.681 1388-1388/? V/EmulatedCamera_Camera: getCameraInfo
2018-08-05 15:34:43.682 1388-1388/? V/EmulatedCamera_BaseCamera: getCameraInfo
2018-08-05 15:34:43.682 1388-2370/? I/CameraService: CameraService::connect call (PID -1 "com.some.packet", camera ID 1) for HAL version default and Camera API version 1
2018-08-05 15:34:43.731 18137-18137/com.some.packet I/CameraCallbacks: onDeviceOrientationChanged 0
2018-08-05 15:34:43.732 18137-18137/com.some.packet V/CameraCallbacks: onDeviceOrientationChanged 0
2018-08-05 15:34:43.732 18137-18137/com.some.packet V/CameraCallbacks: onDeviceOrientationChanged 0
2018-08-05 15:34:43.732 18137-18137/com.some.packet I/Choreographer: Skipped 69 frames!  The application may be doing too much work on its main thread.
2018-08-05 15:34:43.741 18137-18137/com.some.packet W/CameraView: preview: isHardwareAccelerated: true
2018-08-05 15:34:43.741 18137-18137/com.some.packet V/CameraView: preview: isHardwareAccelerated: true
2018-08-05 15:34:43.741 18137-18137/com.some.packet V/CameraView: preview: isHardwareAccelerated: true
2018-08-05 15:34:43.746 18137-18137/com.some.packet W/CameraView: preview: isHardwareAccelerated: true
2018-08-05 15:34:43.746 18137-18137/com.some.packet V/CameraView: preview: isHardwareAccelerated: true
2018-08-05 15:34:43.746 18137-18137/com.some.packet V/CameraView: preview: isHardwareAccelerated: true
[... cut ...]
2018-08-05 15:34:47.446 2599-18207/com.google.android.googlequicksearchbox:search I/MicroRecognitionRunner: Detection finished
2018-08-05 15:34:47.446 2599-18207/com.google.android.googlequicksearchbox:search W/ErrorReporter: reportError [type: 211, code: 524300]: Error reading from input stream
2018-08-05 15:34:47.447 2599-4491/com.google.android.googlequicksearchbox:search I/MicroRecognitionRunner: Stopping hotword detection.
2018-08-05 15:34:47.447 2599-18207/com.google.android.googlequicksearchbox:search W/ErrorProcessor: onFatalError, processing error from engine(4)

KORuL, 2018-08-06:
Something interesting
https://stackoverflow.com/questions/26305107/how-to-fix-fail-to-connect-to-camera-service-exception-in-android-emulator

KORuL, 2018-08-06:
https://github.com/natario1/CameraView/issues/154

natario1, 2018-08-07:
@KORuL can you reproduce this in the demo app?

natario1, 2018-08-07:
The error `CameraService::connect (PID -1) rejected (too many other clients connecting).` and the fact that all other lines are repeated 3 times, makes me think that you are using more than one `CameraView` at once which is **not possible**.

If this is the case I am happy that we do not post this error to the error callback because this is a developer error that you should fix before going in production.

If you can reproduce the error in the demo app, please post all logs from there!

HBiSoft, 2018-08-10:
@KORuL the fact that you implemented the initialization incorrectly doesn't mean that the rest of us did the same. I'm glad you found your issue and it is working on the emulator. 

- I'm only using one instance of `CameraView` (I can't think why someone would even try to use more then one).
- I'm getting this on real devices, not on a emulator (the app is live on Google Play).

As mentioned previously, I can't post a log because I don't have a device to reproduce the issue.

If it is a developer error, like you are saying, then please tell me what I have done that could cause this issue:

```
public class camView extends Activity{
    CameraView cameraView;
    ImageView startRec, stopRec, flipCam;
    String filename;
    File directoryToStore;
    File filePlusName;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        getWindow().setFlags(WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED,
                WindowManager.LayoutParams.FLAG_HARDWARE_ACCELERATED);
        setContentView(R.layout.activity_cam_view);

        cameraView = findViewById(R.id.camera);
        flipCam = findViewById(R.id.flipCam);
        startRec = findViewById(R.id.startRec);
        stopRec = findViewById(R.id.stopRec);

        cameraView.mapGesture(Gesture.PINCH, GestureAction.ZOOM);
        cameraView.mapGesture(Gesture.TAP, GestureAction.FOCUS_WITH_MARKER);
        cameraView.mapGesture(Gesture.LONG_TAP, GestureAction.CAPTURE);

        SimpleDateFormat formatter = new SimpleDateFormat("yyyy_MM_dd_HH_mm_ss");
        Date now = new Date();
        filename = formatter.format(now) + ".mp4";

        directoryToStore = getBaseContext().getExternalFilesDir("CameraTemp");

        filePlusName = new File(directoryToStore, filename);

        flipCam.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                if(getApplicationContext().getPackageManager().hasSystemFeature(PackageManager.FEATURE_CAMERA_FRONT)) {
                    if (cameraView.getFacing() == Facing.FRONT) {
                        cameraView.setFacing(Facing.BACK);
                    } else {
                        cameraView.setFacing(Facing.FRONT);
                    }
                }else {
                    Toast.makeText(camView.this, "Your device doesn't have a front camera..", Toast.LENGTH_SHORT).show();
                }

            }
        });

        if (startRec.getVisibility() == View.VISIBLE) {
            startRec.setOnClickListener(new View.OnClickListener() {
                @Override
                public void onClick(View v) {
                    startRec.setVisibility(View.GONE);
                    stopRec.setVisibility(View.VISIBLE);
                    startCaptureVideo();
                }

            });
        }
        if (startRec.getVisibility() == View.VISIBLE) {
            stopRec.setOnClickListener(new View.OnClickListener() {
                @Override
                public void onClick(View v) {
                    stopRec.setVisibility(View.GONE);
                    startRec.setVisibility(View.VISIBLE);
                    stopCaptureVideo();
                }
            });

        }
    }

    private void startCaptureVideo() {
        cameraView.startCapturingVideo(filePlusName);

    }

    private void stopCaptureVideo() {
        cameraView.stopCapturingVideo();
        camListener();
    }

    private void camListener() {
        cameraView.addCameraListener(new CameraListener() {
            @Override
            public void onCameraOpened(CameraOptions options) {

            }

            @Override
            public void onVideoTaken(File video) {
                String path = filePlusName.getAbsolutePath();

                Intent intent = new Intent(camView.this, VideoActivity.class);
                intent.putExtra("videoUri", path);
                intent.setData(FileProvider.getUriForFile(getApplication(),getApplication().getPackageName() + ".provider", video));
                intent.addFlags(Intent.FLAG_GRANT_READ_URI_PERMISSION);

                startActivity(intent);
                finish();

            }

        });

    }

    @Override
    protected void onResume() {
        super.onResume();
        cameraView.start();
    }

    @Override
    protected void onPause() {
        super.onPause();
        cameraView.stop();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        cameraView.destroy();
    }

}
```

natario1, 2018-08-10:
@HBiSoft can you set up logging for your next release? Link your app to something like crashlytics and use CameraLogger.setLogLevel (use verbose to get more logs) and registerLogger to intercept logs and send them to crashlytics.

Then post here all the logs from a session that crashed (from start to crash). You could set up this only for the devices that have this issue (if you know it is device specific) in order not to have useless log entries.

yxunknown, 2018-08-14:
same issue on nubia z17 mini,

natario1, 2018-09-14:
With #265 you can now catch this exception in the error handler, instead of having the app crash.

I still think it would have been cool to understand if there is a bug in the library or not - and this will be more difficult with #265 because we'll get less reports. But we were not able to fix this so...

I will close this now. If you get this exception again (after #265 is merged), please open a new issue with CameraLogger logs so we can see if there's something wrong on our side.

closed 2018-09-14 by natario1

KingBright, 2018-09-17:
This issue occurs when the camera is in a state that is not ready for use, the cause may be that some other application(also maybe your own code) did not release the camera correctly after they used it. For this situation, you should find a solution to recover(not limited in coding way, maybe you need a reboot), and just give an alert to users.

HBiSoft, 2019-05-18:
@natario1 Log as you requested above:

    1 | \| | 05:10:02:131 (UTC) | \| | ?/CameraController Start: posting runnable. State: STATE_STOPPED
    2 | \| | 05:10:02:132 (UTC) | \| | ?/CameraController Start: executing. State: STATE_STOPPED
    3 | \| | 05:10:02:133 (UTC) | \| | ?/CameraController Start: about to call onStart() STATE_STARTING
    4 | \| | 05:10:02:134 (UTC) | \| | ?/Camera1 collectCameraId Facing: BACK Internal: 0 Cameras: 2
    5 | \| | 05:10:02:169 (UTC) | \| | ?/Camera1 createCamera: Applying default parameters.
    6 | \| | 05:10:02:186 (UTC) | \| | V/CameraView preview: isHardwareAccelerated: true
    7 | \| | 05:10:02:195 (UTC) | \| | ?/Camera1 onStart: Ended
    8 | \| | 05:10:02:196 (UTC) | \| | ?/CameraController Start: returned from onStart(). Dispatching. STATE_STARTING
    9 | \| | 05:10:02:196 (UTC) | \| | ?/CameraCallbacks dispatchOnCameraOpened com.otaliastudios.cameraview.CameraOptions@580bbaf
    10 | \| | 05:10:02:201 (UTC) | \| | V/CameraView onMeasure: surface is not ready. Calling default behavior.
    11 | \| | 05:10:02:202 (UTC) | \| | V/CameraView onMeasure: surface is not ready. Calling default behavior.
    12 | \| | 05:10:02:223 (UTC) | \| | V/CameraView onMeasure: surface is not ready. Calling default behavior.
    13 | \| | 05:10:02:224 (UTC) | \| | V/CameraView onMeasure: surface is not ready. Calling default behavior.
    14 | \| | 05:10:02:252 (UTC) | \| | ?/CameraPreview dispatchOnOutputSurfaceAvailable: w= 1080 h= 1920
    15 | \| | 05:10:02:253 (UTC) | \| | ?/Camera1 onSurfaceAvailable: Size is 1080x1920
    16 | \| | 05:10:02:254 (UTC) | \| | ?/Camera1 onSurfaceAvailable: Inside handler. About to bind.
    17 | \| | 05:10:02:255 (UTC) | \| | ?/Camera1 bindToSurface: Started
    18 | \| | 05:10:02:258 (UTC) | \| | ?/CameraController computeCaptureSize: result: 1080x1920 flip: true mode: VIDEO
    19 | \| | 05:10:02:265 (UTC) | \| | ?/Camera1 size: sizesFromList: [1920x1080, 1440x1080, 1072x1072, 1280x720, 1056x704, 960x720, 880x720, 720x720, 720x480, 640x480, 352x288, 320x240, 176x144]
    20 | \| | 05:10:02:267 (UTC) | \| | ?/CameraController size: computePreviewSize: targetRatio: 9:16 targetMinSize: 1080x1920
    21 | \| | 05:10:02:269 (UTC) | \| | ?/CameraController computePreviewSize: result: 1920x1080 flip: true
    22 | \| | 05:10:02:269 (UTC) | \| | ?/Camera1 onSurfaceAvailable Dispatching onCameraPreviewSizeChanged.
    23 | \| | 05:10:02:270 (UTC) | \| | ?/CameraCallbacks onCameraPreviewSizeChanged
    24 | \| | 05:10:02:271 (UTC) | \| | ?/CameraPreview setInputStreamSize: desiredW= 1080 desiredH= 1920
    25 | \| | 05:10:02:278 (UTC) | \| | ?/CameraController computeCaptureSize: result: 3096x4128 flip: true mode: PICTURE
    26 | \| | 05:10:02:287 (UTC) | \| | ?/Camera1 onSurfaceAvailable Starting preview with startPreview().
    27 | \| | 05:10:02:319 (UTC) | \| | ?/CameraView onMeasure: requested dimensions are (1080[EXACTLY]x1920[EXACTLY])
    28 | \| | 05:10:02:322 (UTC) | \| | ?/CameraView onMeasure: previewSize is (1080.0x1920.0)
    29 | \| | 05:10:02:323 (UTC) | \| | V/CameraView onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1920)
    30 | \| | 05:10:02:324 (UTC) | \| | ?/CameraView onMeasure: requested dimensions are (1080[EXACTLY]x1920[EXACTLY])
    31 | \| | 05:10:02:325 (UTC) | \| | ?/CameraView onMeasure: previewSize is (1080.0x1920.0)
    32 | \| | 05:10:02:326 (UTC) | \| | V/CameraView onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1920)
    33 | \| | 05:10:02:370 (UTC) | \| | ?/CameraView onMeasure: requested dimensions are (1080[EXACTLY]x1920[EXACTLY])
    34 | \| | 05:10:02:371 (UTC) | \| | ?/CameraView onMeasure: previewSize is (1080.0x1920.0)
    35 | \| | 05:10:02:372 (UTC) | \| | V/CameraView onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1920)
    36 | \| | 05:10:02:373 (UTC) | \| | ?/CameraView onMeasure: requested dimensions are (1080[EXACTLY]x1920[EXACTLY])
    37 | \| | 05:10:02:374 (UTC) | \| | ?/CameraView onMeasure: previewSize is (1080.0x1920.0)
    38 | \| | 05:10:02:375 (UTC) | \| | V/CameraView onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (1080x1920)
    39 | \| | 05:10:02:736 (UTC) | \| | ?/Camera1 onSurfaceAvailable Started preview.
    40 | \| | 05:10:06:942 (UTC) | \| | ?/Camera1 stopVideo mVideoRecorder is null? false
    41 | \| | 05:10:07:204 (UTC) | \| | V/FullVideoRecorder stop: Error while closing media recorder. Swallowing java.lang.RuntimeException: stop failed.
    42 | \| | 05:10:07:211 (UTC) | \| | ?/CameraCallbacks dispatchError com.otaliastudios.cameraview.CameraException

alifarina, 2019-09-23:
I have similar code as @HBiSoft . I have already wasted two days trying to fix it. If anyone has a solution please let me know.
Thanks.

HBiSoft, 2019-09-23:
@alifarina What device are you using and what version of `CameraView` are you using? Also, when does the crash happen? Mine happened as soon as I start the recording.

alifarina, 2019-09-23:
Hi
I have OnePlus 6T and I am not using camera view. I am using a camera
intent to open the device default camera. Sometimes it crashes just after
the camera is open and sometimes it crashes after taking a picture.

The error log doesn't say much, just that the process was killed due to low
memory.

HBiSoft, 2019-11-21:
@natario1 Here is the log as you asked for above:

```txt
------ exception.getReason() started -----

1

------ exception.getReason() ended ------

------ exception.toString() started ------

com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service

------ exception.toString() ended ------

------ exception.getMessage() started ------

java.lang.RuntimeException: Fail to connect to camera service

------ exception.getMessage() ended ------

------ exceptionStackTrace started ------

com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service
	at com.otaliastudios.cameraview.engine.Camera1Engine.g(Camera1Engine.java:13)
	at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:3)
	at com.otaliastudios.cameraview.engine.CameraEngine$3.call(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:4)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:1)
	at com.google.android.gms.tasks.zzf.run(Unknown Source)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:2)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:1)
	at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
	at com.google.android.gms.tasks.zzr.zza(Unknown Source)
	at com.google.android.gms.tasks.zzu.zze(Unknown Source)
	at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:7)
	at com.otaliastudios.cameraview.engine.CameraEngine.startEngine(CameraEngine.java:2)
	at com.otaliastudios.cameraview.engine.CameraEngine.b(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:2)
	at com.otaliastudios.cameraview.engine.CameraEngine$17$1.call(CameraEngine.java:1)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:4)
	at com.otaliastudios.cameraview.engine.Step$2.then(Step.java:1)
	at com.google.android.gms.tasks.zzf.run(Unknown Source)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler.run(WorkerHandler.java:2)
	at com.otaliastudios.cameraview.internal.utils.WorkerHandler$1.execute(WorkerHandler.java:1)
	at com.google.android.gms.tasks.zze.onComplete(Unknown Source)
	at com.google.android.gms.tasks.zzr.zza(Unknown Source)
	at com.google.android.gms.tasks.zzu.zze(Unknown Source)
	at com.google.android.gms.tasks.zzu.continueWithTask(Unknown Source)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:7)
	at com.otaliastudios.cameraview.engine.Step.a(Step.java:5)
	at com.otaliastudios.cameraview.engine.CameraEngine$17.run(CameraEngine.java:2)
	at android.os.Handler.handleCallback(Handler.java:733)
	at android.os.Handler.dispatchMessage(Handler.java:95)
	at android.os.Looper.loop(Looper.java:157)
	at android.os.HandlerThread.run(HandlerThread.java:61)
Caused by: java.lang.RuntimeException: Fail to connect to camera service
	at android.hardware.Camera.native_setup(Native Method)
	at android.hardware.Camera.<init>(Camera.java:420)
	at android.hardware.Camera.open(Camera.java:372)
	at com.otaliastudios.cameraview.engine.Camera1Engine.g(Camera1Engine.java:1)
	... 32 more

------ exceptionStackTrace ended ------

```