natario1/CameraView #191: java.lang.NoClassDefFoundError: Failed resolution of: Landroid/support/media/ExifInterface;

OLDdriver2, 2018-04-01:
- [x ] *I am using the latest version*
- [x ] *I can reproduce the bug in the demo app, and if not...*
- [x ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.Target platform Lolipop 5.0
2.SDK API level 26
3.capturePicture()
#### Expected behavior:
Take a picture
#### Actual behavior:
Crashed
#### Screenshots:

#### Interesting logs:
`04-01 07:36:23.805 7945-7989/com.example.cqedu.lpr E/AndroidRuntime: FATAL EXCEPTION: FallbackCameraThread
                                                                     Process: com.example.cqedu.lpr, PID: 7945
                                                                     java.lang.NoClassDefFoundError: Failed resolution of: Landroid/support/media/ExifInterface;
                                                                         at com.otaliastudios.cameraview.CameraUtils.decodeBitmap(CameraUtils.java:114)
                                                                         at com.otaliastudios.cameraview.CameraUtils$1.run(CameraUtils.java:91)
                                                                         at android.os.Handler.handleCallback(Handler.java:819)
                                                                         at android.os.Handler.dispatchMessage(Handler.java:99)
                                                                         at android.os.Looper.loop(Looper.java:164)
                                                                         at android.os.HandlerThread.run(HandlerThread.java:65)
                                                                      Caused by: java.lang.ClassNotFoundException: Didn't find class "android.support.media.ExifInterface" on path: DexPathList[[zip file "/data/app/com.example.cqedu.lpr-dewriNcKuVVlUrYGUKsoGg==/base.apk"],nativeLibraryDirectories=[/data/app/com.example.cqedu.lpr-dewriNcKuVVlUrYGUKsoGg==/lib/x86, /system/lib]]
                                                                         at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:125)
                                                                         at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
                                                                         at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
                                                                         at com.otaliastudios.cameraview.CameraUtils.decodeBitmap(CameraUtils.java:114) 
                                                                         at com.otaliastudios.cameraview.CameraUtils$1.run(CameraUtils.java:91) 
                                                                         at android.os.Handler.handleCallback(Handler.java:819) 
                                                                         at android.os.Handler.dispatchMessage(Handler.java:99) 
                                                                         at android.os.Looper.loop(Looper.java:164) 
                                                                         at android.os.HandlerThread.run(HandlerThread.java:65) `
> 
> `import android.support.media.ExifInterface;`
> This line in CameraUtils.java:11 seems to be wrong.It cannot find the "media" class. 
> `import android.media.ExifInterface;`
> This statement seems to be right.
> So, is it a SDK version problem?And which version should i use?
*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

Patte1808, 2018-04-05:
I was able to solve this issue in my app by adding 'com.android.support:exifinterface:${YOUR_SUPPORT_LIBRARY_VERSION}' into my app build.gradle

natario1, 2018-04-07:
There might be something weird in your build file, because we are using the support ExifInterface library and the class path is correct: https://developer.android.com/reference/android/support/media/ExifInterface.html

hordurhauks, 2018-04-11:
You need to add 

implementation "com.android.support:exifinterface:$supportLibraryVersion" 

into dependencies in your build.gradle file. Note: The $supportLibraryVersion variable is defined outside dependencies like this 

ext.supportLibraryVersion = '27.1.1'

DmytroBatyuk, 2019-02-15:
`implementation 'androidx.exifinterface:exifinterface:1.0.0'` if use android x

PR #222 (2018-05-30): Bump version
- Fixes #180 
- Fixes #191 
- Update to 1.5.1