natario1/CameraView #192: Fatal Exception: java.lang.RuntimeException startPreview failed

chiyan171, 2018-04-02:
Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera._startPreview(Camera.java)
       at android.hardware.Camera.startPreview(Camera.java:823)
       at com.otaliastudios.cameraview.Camera1$12$2.onPictureTaken(Camera1.java:521)
       at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1193)
       at android.os.Handler.dispatchMessage(Handler.java:102)
       at android.os.Looper.loop(Looper.java:135)
       at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2018-04-07:
I think this is not different from #122 , please follow up there!

closed 2018-04-07 by natario1

wider2, 2018-07-20:
E/UncaughtException: java.lang.RuntimeException: Fail to connect to camera service
        at android.hardware.Camera.<init>(Camera.java:545)
        at android.hardware.Camera.open(Camera.java:385)
        at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:172)
        at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:180)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:135)
        at android.os.HandlerThread.run(HandlerThread.java:61)