natario1/CameraView #195: Error don't have callback

ErickDaniel, 2018-04-04:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

Issue: no callback to error
Class: Camera1

When code fails on
	Line: 586
		Camera1.this.mMediaRecorder.start();

Call
	Line: 591
		Camera1.this.endVideoImmediately();

But when code reach
	Line: 616
	this.mMediaRecorder.stop();

Fail again and don't have callback and only log the error

In the commit the update is
	Line: 619
		this.mCameraCallbacks.dispatchError(new CameraException(var2));

Error in log:
`04-04 17:21:12.620 21520-21570/mx.suramexico.suraevolucion E/MediaRecorder: start failed: -38
04-04 17:21:12.622 21520-21570/mx.suramexico.suraevolucion E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.IllegalStateException
                                                                      java.lang.IllegalStateException
                                                                          at android.media.MediaRecorder._start(Native Method)
                                                                          at android.media.MediaRecorder.start(MediaRecorder.java:1170)
                                                                          at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:632)
                                                                          at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
                                                                          at android.os.Handler.handleCallback(Handler.java:751)
                                                                          at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                          at android.os.Looper.loop(Looper.java:154)
                                                                          at android.os.HandlerThread.run(HandlerThread.java:61)
04-04 17:21:12.623 21520-21570/mx.suramexico.suraevolucion I/Camera1: endVideoImmediately: is capturing: true
04-04 17:21:12.624 21520-21570/mx.suramexico.suraevolucion I/MediaRecorderJNI: stop
04-04 17:21:12.624 21520-21570/mx.suramexico.suraevolucion E/MediaRecorder: stop called in an invalid state: 0
04-04 17:21:12.625 21520-21570/mx.suramexico.suraevolucion W/Camera1: endVideoImmediately: Error while closing media recorder. Swallowing java.lang.IllegalStateException
                                                                      java.lang.IllegalStateException
                                                                          at android.media.MediaRecorder._stop(Native Method)
                                                                          at android.media.MediaRecorder.stop(MediaRecorder.java:1199)
                                                                          at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:662)
                                                                          at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:26)
                                                                          at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:637)
                                                                          at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
                                                                          at android.os.Handler.handleCallback(Handler.java:751)
                                                                          at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                          at android.os.Looper.loop(Looper.java:154)
                                                                          at android.os.HandlerThread.run(HandlerThread.java:61)
04-04 17:21:12.626 21520-21570/mx.suramexico.suraevolucion I/CameraCallbacks: dispatchError com.otaliastudios.cameraview.CameraException: java.lang.IllegalStateException
                                                                              com.otaliastudios.cameraview.CameraException: java.lang.IllegalStateException
                                                                                  at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:666)
                                                                                  at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:26)
                                                                                  at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:637)
                                                                                  at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62)
                                                                                  at android.os.Handler.handleCallback(Handler.java:751)
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:95)
                                                                                  at android.os.Looper.loop(Looper.java:154)
                                                                                  at android.os.HandlerThread.run(HandlerThread.java:61)
                                                                               Caused by: java.lang.IllegalStateException
                                                                                  at android.media.MediaRecorder._stop(Native Method)
                                                                                  at android.media.MediaRecorder.stop(MediaRecorder.java:1199)
                                                                                  at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:662)
                                                                                  at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:26) 
                                                                                  at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:637) 
                                                                                  at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:62) 
                                                                                  at android.os.Handler.handleCallback(Handler.java:751) 
                                                                                  at android.os.Handler.dispatchMessage(Handler.java:95) 
                                                                                  at android.os.Looper.loop(Looper.java:154) 
                                                                                  at android.os.HandlerThread.run(HandlerThread.java:61) 
04-04 17:21:12.626 21520-21570/mx.suramexico.suraevolucion I/MediaRecorderJNI: release
`

ErickDaniel, 2018-04-04:
I Fork the project and add the next line
`mCameraCallbacks.dispatchError(new CameraException(e));`

check it and let me know if I can do pull request

Regards

natario1, 2018-04-07:
Instead of passing this to the error callback, we would like to fix the error so it does not occur anymore. Thanks for the stack trace , if you have more info please add here

natario1, 2018-04-07:
I think this could happen during configuration changes / activity destruction. In that case, it's intended behavior. We stop the recorder 'twice' in some conditions, and we ignore if it fails.

So please tell me, is this an app dysfunction (error, crash) or just a log that bothers you in logcat?

ErickDaniel, 2018-04-09:
Well when I launch the camera, the screen shows the camera preview and need start the camera again to rec

MaxMaes, 2018-04-11:
I think i'm experiencing the same problem. When I'm recording a video, I sometimes get a callback to OnVideoTaken without me calling stopCapturingVideo and the video I end up with is unplayable.

WillemTax, 2018-04-11:
@natario1 , same issue as @MaxMaes here.
So we experience a crash, not only a log that bothers us.

MaxMaes, 2018-04-11:
I seem to be able to reproduce the issue pretty consistently by doing the following:

- When the app starts, start recording immediatly
- Shake the phone while recording
- After a second or two the recording fails

tuabali, 2022-08-30:
Kindly check your camera or mic already in use by your app or other app.