natario1/CameraView #197: Frame processor error

huweijian5, 2018-04-10:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ 1.5.0] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.size:480*320
2.
3.

#### Expected behavior:
frame data size is 230400
#### Actual behavior:
230401
#### Screenshots:
[image]

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

ssakhavi, 2018-04-19:
I also have encountered the same problem.

natario1, 2018-04-24:
I think we can remove the +1 here

https://github.com/natario1/CameraView/blob/f6b2506ecf755527c93ba419268797a033a785c4/cameraview/src/main/java/com/otaliastudios/cameraview/FrameManager.java#L91

Can some of you submit a PR?

natario1, 2018-04-25:
Thanks @ssakhavi !

closed 2018-04-25 by natario1