natario1/CameraView #203: The result image are rotated.

jinfagang, 2018-04-22:
Just using the default settings, the result of images are rotated. How to set it as not rotated anyway?

canuludag, 2018-04-25:
Hi @jinfagang ,
You should use `CameraUtils.decodeBitmap(` method. Camera response `byte[]` data contains image rotation info. And the resulted bitmap conversion works as expected with correct orientation. 
Do not use `Bitmap cameraBitmap = BitmapFactory.decodeByteArray(byteArray, 0, byteArray.length);` bitmap creation method. Instead use this one:

```
CameraUtils.decodeBitmap(byteArray, new CameraUtils.BitmapCallback() {
            @Override
            public void onBitmapReady(Bitmap bitmap) {
                // Do whatever you want with your Bitmap
                // Save it as file or set into ImageView
            }
        });
```
Hope this helps!

jinfagang, 2018-04-26:
Thanks, already figure it out.

MoustafaElsaghier, 2018-05-06:
this solution does not work with me.

when I capture an image in landscape mode (in forced portrait activity ) it rotates the image to portrait.

jinfagang, 2018-05-07:
@MoustafaElsaghier  This happend specific on some model Phones. You can judge it and rotate the matrix.

MoustafaElsaghier, 2018-05-07:
@jinfagang yeah I did it and worked well by getting value if phone rotated by and rotate the matrix with it.

MoustafaElsaghier, 2018-05-12:
@jinfagang hello, sorry for asking again. will if differ from front and back camera?

s It works well for me in front camera only and for back camera portrait mode get rotated by 180

closed 2018-05-30 by natario1