natario1/CameraView #215: Camera is stretched after clicking picture and it takes time to take picture.

GrishmaPromact, 2018-05-17:
I am using version 1.5.0.
I am using  camera.capturePicture(); method to click picture.
After taking picture camera is stretched for some time and it takes time to take picture.

One more bug is also that i have facing.
if i am using camera.captureSnapshot(); method to take picture then auto flash is not working.
 camera.captureSnapshot(); for this method camera is not getting stretched and image taken is very faster.That is good. But in that method auto flash is not working.Also picture quality is low.

I need quality and also faster image taking without stretching and legging .
Can you help me to solve one of the bug?

It would be great help for me as i am using your library in my project,.

PFA, of my cameraActivity and it's xml file. That would help you to solve my issue.

Thanks in advance.

[VisualogyxCameraActivity.docx](https://github.com/natario1/CameraView/files/2013202/VisualogyxCameraActivity.docx)

natario1, 2018-05-30:
Hi. That's how the lib work, either you use capturePicture() for full functionality or captureSnapshot() for quick snapshot.

You can reduce `capturePicture()` time by choosing a small capture size and avoiding `cropOutput`.

About stretching there is already a bug opened out there but I don't think there is anything we can do.

GrishmaPromact, 2018-05-30:
Hello, But _capturePicture()_ method takes time to take picture. And if i use _captureSnapshot()_ then flash are not working.

natario1, 2018-05-30:
I know... that's how it works!

GrishmaPromact, 2018-05-30:
But i think you should provide flash while taking snapshots.