natario1/CameraView #218: How to use two Camera View in single layout/

benevolent13, 2018-05-29:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.
2.
3.

#### Expected behavior:

#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

benevolent13, 2018-05-29:
E/Camera1: Error inside the onError callback. 2
05-29 12:54:00.366 23461-24845/com.hitesh.camerapreview E/CameraController: uncaughtException: Interrupting thread with state: STATE_STARTED due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 2
    com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Error inside the onError callback. 2
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:272)
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1334)
        at android.os.Handler.dispatchMessage(Handler.java:111)
        at android.os.Looper.loop(Looper.java:207)
        at android.os.HandlerThread.run(HandlerThread.java:61)
     Caused by: java.lang.RuntimeException: Error inside the onError callback. 2
        at com.otaliastudios.cameraview.Camera1.onError(Camera1.java:272) 
        at android.hardware.Camera$EventHandler.handleMessage(Camera.java:1334) 
        at android.os.Handler.dispatchMessage(Handler.java:111) 
        at android.os.Looper.loop(Looper.java:207) 
        at android.os.HandlerThread.run(HandlerThread.java:61)

natario1, 2018-05-30:
Not possible!

Namazed, 2018-06-25:
Hi, @benevolent13 
As already written @natario1 it's not possible, because we may only have one open camera in the process.

closed 2018-07-31 by natario1

toktogul, 2019-02-13:
but in some devices it works. How can I detect, which one works, and which one does not works ?