natario1/CameraView #219: Solution for getting correct image in front cam (Unmirrored image)

vinayakpurohit, 2018-05-29:
So I came up with a solution for getting the image correctly in front cam i.e. the mirrored image. So here is the code.

Use the below function to rotate the image.

 public static byte[] rotateImage(byte[] source) {
        Bitmap bmp = BitmapFactory.decodeByteArray(source, 0, source.length);
        Matrix matrix = new Matrix();
        matrix.preScale(-1, 1);
        Bitmap m = Bitmap.createBitmap(bmp, 0, 0, bmp.getWidth(), bmp.getHeight(), matrix, true);
        ByteArrayOutputStream stream = new ByteArrayOutputStream();
        m.compress(Bitmap.CompressFormat.JPEG, 100, stream);
        byte[] flippedImageByteArray = stream.toByteArray();
        return flippedImageByteArray;

    }

Use this below function to use the images from both facing wisely

cameraView.addCameraListener(new CameraListener() {
            @Override
            public void onPictureTaken(byte[] picture) {
                Intent i = new Intent(getActivity(), ShowImageViewActivity.class);
                if(cameraView.getFacing() == Facing.FRONT){
                    byte[] image = rotateImage(picture);
                    i.putExtra("image",image);
                }else {
                    i.putExtra("image",picture);
                }
                startActivity(i);
            }
        });

inmeriel25, 2018-05-30:
very helpful for rotating images
thanks

natario1, 2018-05-30:
There is some discussion about this in #15 . Basically it is expensive and we don't want to do this in the main lib. But you can do this on your own as you did 👍

closed 2018-09-18 by natario1

SwiiZe, 2018-10-29:
It does not work in using real time Frame.
Why it comes to:  Bitmap bmp = BitmapFactory.decodeByteArray(source, 0, source.length);
Bmp is null. I don't know why

AkshayDandale, 2019-09-23:
How to receive the bitmap on other Activity then?