natario1/CameraView #227: Frame preview lags if frame is held for long inside frame processor

chittorakalpit, 2018-06-06:
Frame preview lags if ```frame``` is held for long inside frame processor.

This happens even if I just freeze the frame and give it to another variable.
```frame``` is being released after the operation by calling ```frame.release()```

Solution:
Use multiple buffers for the camera.

chittorakalpit, 2018-06-06:
Okay so now a new issue has come up.

The Frame processor stops coming to the frame Processor's process method and my other thread keeps waiting for the new frame and thus is stuck.

Please see into it and provide a solution for it.

natario1, 2018-06-06:
If you are slow at processing frames, you will get less calls, by design. If you need more help you can post your code.

Note that freezing is discouraged, because it copies the buffer in a new buffer.

chittorakalpit, 2018-06-07:
Actually I need to pass the image into a image recognition model. I only pass the latest frame available(not all frames are passed). So I can't do that in the process function.

closed 2018-07-31 by natario1