natario1/CameraView #229: Get Picture orientation inside onPictureTaken method?

DenisDavidek, 2018-06-11:
Hi guys,

first  of all thank you for this amazing library. I was searching and wondering if there is a possibility to get captured picture orientation inside onPictureTaken method. I have heard something about CameraUtils class, but I haven't found an example of using that class.

So is there a way to get an orientation from a captured image? If yes, can you please provide a code sample?

Thank you in advance for any answer.

natario1, 2018-06-11:
Thanks to you.

CameraUtils will give you a correctly rotated `Bitmap`, if that's what you want.
If you want to know the actual rotation value, we offer no utility. But you can do it, take a look at `CameraUtils` source code.

Will add this in v2.

DenisDavidek, 2018-06-12:
Hi natario1,

Here is the current code I am using:

 camera.addCameraListener(object: CameraListener(){
            override fun onPictureTaken(jpeg: ByteArray?) {
                super.onPictureTaken(jpeg)

                val intent = Intent(applicationContext, PreviewActivity::class.java)
                var bitmap : Bitmap

                if(camera.getFacing() == Facing.FRONT) {
                    val image = rotateImage(jpeg)

                    bitmap = BitmapFactory.decodeByteArray(image,0,image.size)
                    val imagePath = createFileInternal(bitmap)
                    intent.putExtra("test", imagePath)

///Experimental part - does not  work
                    val exifInterface = android.support.media.ExifInterface(imagePath)
                    val degree = exifInterface.getAttributeInt(android.support.media.ExifInterface.TAG_ORIENTATION,android.support.media.ExifInterface.ORIENTATION_NORMAL)
                    Log.e("DEGREE ", degree.toString())
/// End of Experimental part
                }
                else{
                    if (jpeg != null) {
                        bitmap = BitmapFactory.decodeByteArray(jpeg, 0, jpeg.size)
                        intent.putExtra("test", createFileInternal(bitmap))
                    }
                }
                intent.putExtra(getString(R.string.key_camera_photo_frame_url), photoFrameUrl)
                Log.e("CAMERA PREVIEW ", "DONE")
                startActivity(intent)

            }

And based on what you said I tried to do a CameraUtils version:

        val callback = object : CameraUtils.BitmapCallback{
            override fun onBitmapReady(bitmap: Bitmap?) {

                val imagePath = createFileInternal(bitmap)
                val exifInterface = ExifInterface(imagePath)
                val degree = exifInterface.getAttributeInt(ExifInterface.TAG_ORIENTATION,ExifInterface.ORIENTATION_NORMAL)
                Log.e("DEGREE ", degree.toString())
            }

        }
        CameraUtils.decodeBitmap(jpeg,callback)

I have put that code inside onPictureTaken method. Unfortunately exif interface is still returning 0 no matter the orientation. Any idea why, please?

Namazed, 2018-06-25:
Hi, @DenisDavidek 
if you want very fast solution for getting orientation without ExifInterface and creating file, you need looking at Telegram source code (CameraController.java/getOrientation method). This solution get orientation from byte array. Sorry, but Im not inserting the code here, because it's big.