natario1/CameraView #232: Transition Animation (CameraView)

ateymoori, 2018-06-23:
Hi
is there any solution to transition animation cameraView to other Activity or fragment? (shared element effect from origin cameraview to destination cameraview) ?
thanks

Namazed, 2018-06-25:
Hi, @ateymoori 
I think this is impossible, because you need close camera when you go out from activity or fragment.
If you don't this, you will have memory leak and some troubles with camera. 
But you may create snapshot and shared it or use shared transition on only one screen (activity or fragment).
Sorry for my English.

ateymoori, 2018-06-28:
thank you so much
I`v tested by snapshot and transition that to new activity but it`s not smooth .
Do you have any other solution?

Namazed, 2018-06-28:
@ateymoori 
It's my pleasure :)
You will can use second way.
Use View which can expand to full screen. This is work fine, because we don't need go out from activity or fragment and don't need open camera two times. 
Maybe you will come up with another way.

ateymoori, 2018-07-02:
thanks
but the view is inside a horizontal RecyclerView
and i cant scale up a view inside it to fill activity 
I am looking for a way to open camera like Telegram camera opening

Namazed, 2018-07-02:
@ateymoori 
Telegram doing this as I wrote above :)

ateymoori, 2018-07-02:
Telegram works so smooth and great in all functionalitis
great product from Russia, Your country :)
thanks from Iran to Russia !  👍  :)
I will test that

Namazed, 2018-07-02:
@ateymoori 
Not at all.
I was happy to help you :)

natario1, 2018-07-31:
Thanks @Namazed for addressing some issues here.

closed 2018-07-31 by natario1