natario1/CameraView #233: Device specific - using 720p results in a RuntimeException. 480p is working fine.

Patte1808, 2018-06-25:
I'm currently facing this issue on a device. Whenever I set the resolution to `max720p` in the demo, the MediaRecorder crashes. Using `max480p` works fine.

Model number: D5803
Android version: 6.0.1

If you need anything more from me, let me know.

```
06-25 14:16:44.473 2358-2358/com.otaliastudios.cameraview.demo E/ControlView: curr: MAX_480P new: MAX_720P
06-25 14:16:44.474 2358-2373/com.otaliastudios.cameraview.demo I/CameraController: shouldFlipSizes: displayOffset= 0 sensorOffset= 90
    shouldFlipSizes: sensorToDisplay= 90
    size: computeCaptureSize: videoQuality: MAX_720P targetRatio: 9:16
    computePictureSize: result: 2952x5248 flip: true
06-25 14:16:44.485 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceChanged, size is 720x1024
06-25 14:16:44.487 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: setVideoQuality: captureSize: 5248x2952
    setVideoQuality: previewSize: 720x480
06-25 14:16:44.489 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: size: sizesFromList: [1920x1080, 1600x1200, 1280x720, 960x720, 720x480, 640x480, 480x320, 320x240, 176x144]
06-25 14:16:44.489 2358-2373/com.otaliastudios.cameraview.demo I/CameraController: shouldFlipSizes: displayOffset= 0 sensorOffset= 90
06-25 14:16:44.490 2358-2373/com.otaliastudios.cameraview.demo I/CameraController: shouldFlipSizes: sensorToDisplay= 90
    size: computePreviewSize: targetRatio: 16:9 targetMinSize: 1024x720
    computePreviewSize: result: 1920x1080 flip: true
06-25 14:16:44.490 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceChanged: Computed a new preview size. Going on.
06-25 14:16:44.618 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceChanged: Dispatching onCameraPreviewSizeChanged.
06-25 14:16:44.618 2358-2373/com.otaliastudios.cameraview.demo I/CameraCallbacks: onCameraPreviewSizeChanged
06-25 14:16:44.618 2358-2373/com.otaliastudios.cameraview.demo I/CameraController: shouldFlipSizes: displayOffset= 0 sensorOffset= 90
    shouldFlipSizes: sensorToDisplay= 90
06-25 14:16:44.618 2358-2373/com.otaliastudios.cameraview.demo I/CameraPreview: setDesiredSize: desiredW= 1080 desiredH= 1920
06-25 14:16:44.627 2358-2358/com.otaliastudios.cameraview.demo I/CameraPreview: crop: applied scaleX= 1.0
    crop: applied scaleY= 1.25
06-25 14:16:44.627 2358-2358/com.otaliastudios.cameraview.demo I/CameraController: shouldFlipSizes: displayOffset= 0 sensorOffset= 90
    shouldFlipSizes: sensorToDisplay= 90
06-25 14:16:44.628 2358-2358/com.otaliastudios.cameraview.demo I/CameraView: onMeasure: requested dimensions are (720[EXACTLY]x1024[EXACTLY])
    onMeasure: previewSize is (1080.0x1920.0)
06-25 14:16:44.628 2358-2358/com.otaliastudios.cameraview.demo W/CameraView: onMeasure: both are MATCH_PARENT or fixed value. We adapt. This means CROP_CENTER. (720x1024)
06-25 14:16:44.629 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceChanged: Starting preview with startPreview().
06-25 14:16:44.760 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: onSurfaceChanged: Started preview.
06-25 14:16:44.790 2358-2379/com.otaliastudios.cameraview.demo D/OpenGLRenderer: endAllActiveAnimators on 0xa0009e00 (ListPopupWindow$DropDownListView) with handle 0x9fd98690
06-25 14:16:46.554 2358-2373/com.otaliastudios.cameraview.demo E/MediaRecorder: start failed: -19
06-25 14:16:46.555 2358-2373/com.otaliastudios.cameraview.demo E/Camera1: Error while starting MediaRecorder. Swallowing. java.lang.RuntimeException: start failed.
    java.lang.RuntimeException: start failed.
        at android.media.MediaRecorder.start(Native Method)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:629)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:64)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:234)
        at android.os.HandlerThread.run(HandlerThread.java:61)
06-25 14:16:46.555 2358-2373/com.otaliastudios.cameraview.demo I/Camera1: endVideoImmediately: is capturing: true
06-25 14:16:46.555 2358-2373/com.otaliastudios.cameraview.demo E/ExtendedMediaRecorder: stop called in an invalid state: 0
06-25 14:16:46.558 2358-2373/com.otaliastudios.cameraview.demo W/Camera1: endVideoImmediately: Error while closing media recorder. Swallowing java.lang.IllegalStateException
    java.lang.IllegalStateException
        at android.media.MediaRecorder.stop(Native Method)
        at com.otaliastudios.cameraview.Camera1.endVideoImmediately(Camera1.java:659)
        at com.otaliastudios.cameraview.Camera1.access$1500(Camera1.java:28)
        at com.otaliastudios.cameraview.Camera1$14.run(Camera1.java:634)
        at com.otaliastudios.cameraview.Camera1$2.run(Camera1.java:64)
        at android.os.Handler.handleCallback(Handler.java:739)
        at android.os.Handler.dispatchMessage(Handler.java:95)
        at android.os.Looper.loop(Looper.java:234)
        at android.os.HandlerThread.run(HandlerThread.java:61)

```

Patte1808, 2018-06-25:
I debugged it a little bit.

The 720p profile seems to exist on the device, the problematic place seems to be  this (Camera1.java, line 687):
`mMediaRecorder.setVideoSize(profile.videoFrameWidth, profile.videoFrameHeight);
`

So I wonder: can it happen that the manufacturer lies, or that the profile information can sometimes be wrong? Using a predefined width and height like 640 * 480 there works.

Patte1808, 2018-06-26:
So I think the problem is that sometimes you have to use setProfile and sometimes you have to hardcode the values.

This is at least what I understand by the research I did.
Found those 2 blogs:
https://traxnet.wordpress.com/2011/11/03/android-mediarecorder-with-camera/
http://joerg-richter.fuyosoft.com/?p=127

Namazed, 2018-09-13:
Hi, @Patte1808 
I had this problem. I fixed it in this pull request GH-264

natario1, 2018-09-20:
The PR was merged and published. Feel free to reopen if the issue persists

closed 2018-09-20 by natario1

Patte1808, 2018-10-18:
@Namazed Sorry for the late response, seems to work, thanks for your effort! :)