natario1/CameraView #236: Feature request: A callback method for when time limit is reached

aleksei-tro, 2018-07-02:
First of all - I love the work you're doing on this 👍🏻

Been implementing this into our project and we have a use case where we're enforcing a time limit on the video duration. The user can either let the time run out or stop the recording by pressing a button.
Outside of keeping track of the clicks, there isn't a way for me to know why the recording has stopped.

It would be great to add a method to the `CameraListener.class` class that gets called when a time limit has been reached.

natario1, 2018-07-31:
I would say this is a design flow in the library. The video callback should pass a rich object with metadata, not just the file. But it's not easy to fix without going to a v2 with breaking changes

aleksei-tro, 2018-07-31:
Right.
I just thought that since the library does obey set time limit, when the limit is reached it could trigger the callback.
Sounds like wishful thinking on my part?

natario1, 2018-07-31:
yes, there can be a callback for time limit and size limit. It's just poor developer experience (you have to set a flag, then wait for the video callback.. assuming they come in this order..). But if you want to work on it, we can go for it and remove it in v2

aleksei-tro, 2018-07-31:
Understood. Thanks for getting back to me

closed 2018-09-14 by natario1