natario1/CameraView #241: Bug with tap to focus and Flash

dai-dao, 2018-07-16:
- [x] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Following code
       //
       camera.keepScreenOn = true
       // Pinch to zoom!
        camera.mapGesture(Gesture.PINCH, GestureAction.ZOOM)
        // Tap to focus
        camera.mapGesture(Gesture.TAP, GestureAction.FOCUS_WITH_MARKER)
        // AutoFlash
        cameraView.flash = Flash.AUTO 

#### Expected behavior:
1. when tap in dark place: flash should not be turned on, only the focus
2. when take picture in dark place: the resulting image should be bright

#### Actual behavior:
1. when tap in dark place: flash goes on like crazy
2. when take picture in dark place: the resulting image is dark

#### Screenshots:
Here's a link to the short vid showing what I mean: https://storage.googleapis.com/utils0989/video-1531755323.mp4

#### Interesting logs:

wosiu, 2018-07-30:
I like actual behavior. To my mind it does not make sense to focus without a flash if final photo is going to be with the flash. It's better to reflect final environment during "on tap" focus.

natario1, 2018-09-20:
Unfortunately we are using Camera1 internal flash management so what happens does not depend on our code.

Flash must be turned on during focus for focus to work.

closed 2018-09-20 by natario1