natario1/CameraView #242: Context leaked through SensorManager

MatFl, 2018-07-18:
The OrientationHelper is using the  CameraView's context to create a new OrientationEventListener, which gets the system sensor manager. Android's SystemSensorManager holds the context that was used to create it, and because it is a system service, it acts like a singleton and thus leaks the context.

It would be better to create the OrientationEventListener using the application context, to prevent this leak.

natario1, 2018-07-18:
Thanks, but I don't get it... `and because it is a system service, it acts like a singleton` can you point to AOSP source code showing this?

I don't think Android source code would ever put a context in a static variable.

MatFl, 2018-07-19:
I am not exactly sure where this happens, but we definitely see this leak constantly when using the CameraView in a Fragment. Leakcanary also has some exceptions added for this kind of issue, but only for specific manufactures: https://github.com/square/leakcanary/blob/master/leakcanary-android/src/main/java/com/squareup/leakcanary/AndroidExcludedRefs.java#L502-L510

Anyways, it would not harm to use the application context for the OrientationEventListener.

natario1, 2018-07-21:
And the leak trace points to the orientation listener?

It would do no harm, sure.

MatFl, 2018-07-23:
The leak does not point directly to the listener, but to the Activity, because the activity context is used.
I think this is not an ASOP issue, as we also have niche (enterprise) device, just like the leak canary exception suggests.

The heap dump shows a static hContext field in the sensor manager which is used in the OrientationEventListener. Of course, this is actually an issue of the ROM, but none of us can change that, so using the application context at least prevents the activity from leaking.

[image]

closed 2018-07-31 by natario1