natario1/CameraView #249: Force autofocus when capturePicture() just before picture is taken

wosiu, 2018-07-30:
- [ x] *I am using the latest version*
- [x ] *I can reproduce the bug in the demo app, and if not...*
- [x ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. Turn off flash (Flash.OFF) - important because Flash.AUTO and Flash.ON have its own routines for that
2. move your camera far from some object, hold for 3 sec and let it autofocus
3. move quickly camera close to the object and click photo before it manages to use continuous autofocus
4. That's it - you've got blured image :(

#### Expected behavior:
Proposal number 1)
Just after camera.capturePicture() - autofocus should be called automatically.
Photo should be really taken after autofocus ends. 

Proposal number 2)
add to CameraView API another function:
```public void startAutoFocus(float x, float y, Callback onFocusFinishedCallback);```
so that users can implement their own logic for taking pictures. The idea is to use something like `camera.capturePicture()` as a `onFocusFinishedCallback`, so that we can develop mentioned behaviour by our self.
Why I don't want to override:
```
camera.addCameraListener(new CameraListener() {
    @Override
    public void onFocusEnd(boolean successful, PointF point) { camera.capturePicture(); }
}
```
Well, because it is also called after gesture (screen tap in my case). And I need screen tap only to force autofocus and not take a picture. I've got a button for taking a picture and I wish it to be the only way to do so.

#### Actual behavior:
Proposal number 1)
Autofocus is not called automatically.
Proposal number 2)
Extend API so that developer can implement this behaviour by his own. 

Personally I vote for proposal number 1)

wosiu, 2018-07-30:
Not sure if a bag or enhancement..

amit016verma, 2018-09-06:
this solution does not work for me , I am also want to do same thing Force autofocus when capturePicture() just before picture is taken

natario1, 2018-09-20:
The solution is what you provided:
```

camera.addCameraListener(new CameraListener() {
    @Override
    public void onFocusEnd(boolean successful, PointF point) { 
        camera.removeCameraListener(this);
        camera.capturePicture(); 
    }
}
camera.startAutoFocus(x, y)
```

You can check for successful to be true and for PointF to be equal to your original x/y.

closed 2018-09-20 by natario1