natario1/CameraView #252: How to manage Preview rate

juanjecilla, 2018-08-01:
I am trying to make an app with a real time camera that manage each frame and make some processing. The problem is that I cannot find the way to manage the frame rate the camera provides. How can I manage to process just 1 frame per second for example.

Thanks.

pinodex, 2018-08-13:
What I did as a workaround is to check the last time I processed a frame.

Something like this I used to process 2fps:

```java
private class MyFrameProcessor implements FrameProcessor {

    long lastSend = System.currentTimeMillis();

    @Override
    public void process(Frame frame) {
        long now = System.currentTimeMillis();

        if (now - lastSend < 500) {
            return;
        }

        lastSend = now;

        // Process frame
    }
}
```

juanjecilla, 2018-08-21:
Thanks, I will try!

natario1, 2018-09-19:
Or you can use Thread.sleep(1000) in the process function!

closed 2018-09-19 by natario1