natario1/CameraView #253: Does not work at all Meize M6

KORuL, 2018-08-05:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [x] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. I run on Meizu M6 and get what's on the screenshot
The camera does not open!

#### Expected behavior:
I run on Meizu M6 and get what's on the screenshot
The camera does not open!
#### Actual behavior:

#### Screenshots:
[image]
#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

    --------- beginning of main
08-05 19:08:10.540 415-1194/? I/Netd: Get netlink event, ifname: wlan0 action: 4
08-05 19:08:10.551 1248-1327/system_process E/WifiStateMachine: mIsFullScanOngoing: true, mSendScanResultsBroadcast: true
08-05 19:08:10.551 1248-1327/system_process E/WifiStateMachine: mWifiOnScanCount: 848
08-05 19:08:10.557 2243-5890/? W/ctxmgr: [AclManager]No 2 for (accnt=account#-517948760#, com.google.android.gms(10014):IndoorOutdoorProducer, vrsn=12874000, 0, 3pPkg = null ,  3pMdlId = null ,  pid = 2243). Was: 3 for 57, account#-517948760#
08-05 19:08:10.629 2243-5890/? W/ctxmgr: [AclManager]No 2 for (accnt=account#206561953#, com.google.android.gms(10014):UserLocationProducer, vrsn=12874000, 0, 3pPkg = null ,  3pMdlId = null ,  pid = 2243). Was: 2 for 1, account#206561953#
08-05 19:08:10.632 2243-5890/? W/ctxmgr: [AclManager]No 2 for (accnt=account#1355320518#, com.google.android.gms(10014):UserLocationProducer, vrsn=12874000, 0, 3pPkg = null ,  3pMdlId = null ,  pid = 2243). Was: 2 for 1, account#1355320518#
08-05 19:08:10.658 2243-5891/? I/GCoreUlr: Successfully inserted 1 locations
08-05 19:08:10.675 2243-2250/? I/System: FinalizerDaemon: finalize objects = 602
08-05 19:08:10.784 1248-1327/system_process I/WifiStateMachine: fetchRssiLinkSpeedAndFrequencyNative, newRssi:-25, newLinkSpeed:135, SSID:"HAME"
08-05 19:08:11.280 13784-13784/? E/memtrack: Couldn't load memtrack module (No such file or directory)
08-05 19:08:11.280 13784-13784/? E/android.os.Debug: failed to load memtrack module: -2
08-05 19:08:11.284 13784-13784/? I/Radio-JNI: register_android_hardware_Radio DONE
08-05 19:08:11.316 13784-13784/? I/aoc: System.exit called, status: 0
08-05 19:08:11.316 13784-13784/? I/AndroidRuntime: VM exiting with result code 0.
08-05 19:08:12.622 396-14706/? I/MAL-Daemon:  (        ru_syssig_hndler,  418) [RDS-I][ru_syssig_hndler] Send Event: 688 
08-05 19:08:12.622 396-14706/? I/MAL-Daemon:  (        ru_syssig_hndler,  419) [RDS-I][RDS][RU][ru_syssig_hndler] ru_check_ctx_ptr<0xf60b3c9c, 0xf60b3c9c>
08-05 19:08:12.622 396-14706/? W/MAL-Daemon:  (    mdfx_task_send_event,  636) mdfx_task_send_event(): [ACT] should NOT enter this leagcy API!!
08-05 19:08:12.622 396-14706/? I/MAL-Daemon:  (    ru_send_event_entity,  350) [RDS-I][RDS][RU][ru_send_event_entity] send event 688 to 602 with ret(0)
08-05 19:08:12.622 396-830/? I/MAL-Daemon:  (             rds_ru_task,  445) [RDS-I][RU][rds_ru_task] RCV Event: 688 
08-05 19:08:12.638 396-830/? I/MAL-Daemon:  (  ru_wsl_wifi_status_ind,  734) [RDS-I][RDS][RU][ru_wsl_wifi_status_ind] ru_check_ctx_ptr<0xf60b3c9c, 0xf60b3c9c>
08-05 19:08:12.638 396-830/? W/MAL-Daemon:  (    mdfx_task_send_event,  636) mdfx_task_send_event(): [ACT] should NOT enter this leagcy API!!
08-05 19:08:12.638 396-830/? I/MAL-Daemon:  (    ru_send_event_entity,  350) [RDS-I][RDS][RU][ru_send_event_entity] send event 604 to 602 with ret(0)
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          ru_get_netstat, 5285) [RDS-I][RU][ru_get_netstat] netstat <thrghpt, pkterrt>, wifi <6846, 0%>, mobile <2946, 0%>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (    ru_send_event_entity,  350) [RDS-I][RDS][RU][ru_send_event_entity] send event 614 to 1233 with ret(0)
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (             rds_ru_task,  445) [RDS-I][RU][rds_ru_task] RCV Event: 604 
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (  rds_ru_wifi_status_ind, 1338) [RDS-I][RU][EVENT_RU_WS_WIFI_STATUS_IND] fgwifi_en:1(no use) i4wifirssi:0xffffffe6 i4wifisnr:0x270f
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (    ru_send_event_entity,  350) [RDS-I][RDS][RU][ru_send_event_entity] send event 615 to 1233 with ret(0)
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <0>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (      rds_cell_only_orng, 3608) [RDS-I][RU][RDS IMS/MMS][rds_cell_only_orng] 
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <0>, RAN1 : <254> ,RAN2 : <254>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <1>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (      rds_cell_only_orng, 3608) [RDS-I][RU][RDS IMS/MMS][rds_cell_only_orng] 
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <1>, RAN1 : <254> ,RAN2 : <254>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <2>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (      rds_cell_only_orng, 3608) [RDS-I][RU][RDS IMS/MMS][rds_cell_only_orng] 
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <2>, RAN1 : <254> ,RAN2 : <254>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <3>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <3>, RAN1 : <1> ,RAN2 : <255>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <4>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <4>, RAN1 : <1> ,RAN2 : <255>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4866) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <5>, UI Policy : <3>, RDS Policy : <3>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <5>, RAN1 : <1> ,RAN2 : <255>
08-05 19:08:12.639 396-830/? I/MAL-Daemon:  (          rds_ru_rpl_gen, 4877) [RDS-I][RU][rds_ru_rpl_gen][RDS][OP0] TYPE : <6>, RAN1 : <254> ,RAN2 : <254>
[... 402 lines cut ...]
08-05 19:08:22.220 391-607/? I/SurfaceFlinger: EventThread Client Pid (13826) disconnected by (391)
08-05 19:08:22.221 391-607/? I/SurfaceFlinger: EventThread Client Pid (13826) disconnected by (391)
08-05 19:08:22.230 1248-1260/system_process I/PowerManagerService: releaseWakeLockInternal: lock=102689728, packageName=android, flags=0x0, wakeLock={SCREEN_BRIGHT_WAKE_LOCK        'WindowManager' ON_AFTER_RELEASE (uid=1000, pid=1248, ws=WorkSource{10158})}, sleepTimeout=-1ms, screenOffTimeout=60000ms,screenDimDuration=7000ms
08-05 19:08:22.234 391-391/? I/BufferQueue: [com.otaliastudios.cameraview.demo/com.otaliastudios.cameraview.demo.CameraActivity](this:0x7f85427800,id:332,api:1,p:-1,c:-1) ~BufferQueueCore
08-05 19:08:22.253 391-605/? I/SurfaceFlinger: [SF client] DEL(0x7f84189100) for (1248:system_server)
08-05 19:08:22.255 1248-1354/system_process I/libPerfService: 1: set freq: 1001000, 1508000, 
08-05 19:08:22.258 1248-1354/system_process I/libPerfService: 1: set: -1, 4, 
08-05 19:08:22.258 1248-1354/system_process I/libPerfService: 1: set vcore level: 3
08-05 19:08:22.258 1248-1354/system_process I/libPerfService: perfSetFavorPid - pid:2197, 895
08-05 19:08:22.261 454-454/? I/Zygote: Process 13826 exited cleanly (255)
08-05 19:08:22.279 391-391/? I/BufferQueue: [unnamed-391-333](this:0x7f85427800,id:333,api:0,p:-1,c:-1) BufferQueue core=(391:/system/bin/surfaceflinger)
08-05 19:08:22.283 2197-2605/? W/libEGL: [ANDROID_RECORDABLE] format: 1
08-05 19:08:22.287 2197-2197/? I/System.out: launcher guest mode set draglayout visiable
[... cut ...]
08-05 19:08:22.901 13887-13887/? E/memtrack: Couldn't load memtrack module (No such file or directory)
08-05 19:08:22.902 13887-13887/? E/android.os.Debug: failed to load memtrack module: -2
08-05 19:08:22.905 13887-13887/? I/Radio-JNI: register_android_hardware_Radio DONE
08-05 19:08:22.935 1248-1259/system_process I/ActivityManager: Force stopping com.otaliastudios.cameraview.demo appid=10158 user=0: from pid 13887
08-05 19:08:22.941 1248-1248/system_process I/NotificationService: action=android.intent.action.PACKAGE_RESTARTED queryReplace=false

KORuL, 2018-08-05:
From CameraLogger:

08-05 19:34:51.892 16946-16946/com.some.packed E/CameraLogger: get: Reusing cached worker handler. CameraViewController
08-05 19:34:51.892 16946-16946/com.some.packed E/CameraLogger: get: Reusing cached worker handler. CameraViewWorker
08-05 19:34:51.893 16946-16946/com.some.packed E/CameraLogger: get: Reusing cached worker handler. FrameProcessorsWorker
08-05 19:34:51.912 16946-16946/com.some.packed E/CameraLogger: Start: posting runnable. State: STATE_STOPPED
08-05 19:34:51.912 16946-16946/com.some.packed E/CameraLogger: Start: posting runnable. State: STATE_STOPPED
08-05 19:34:51.913 16946-16946/com.some.packed E/CameraLogger: Stop: posting runnable. State: STATE_STOPPED
08-05 19:34:51.913 16946-16946/com.some.packed E/CameraLogger: Stop: posting runnable. State: STATE_STOPPED
08-05 19:34:51.913 16946-18180/com.some.packed E/CameraLogger: Start: executing. State: STATE_STOPPED
08-05 19:34:51.913 16946-18180/com.some.packed E/CameraLogger: Start: executing. State: STATE_STOPPED
08-05 19:34:51.914 16946-18180/com.some.packed E/CameraLogger: Start: about to call onStart() STATE_STARTING
08-05 19:34:51.914 16946-18180/com.some.packed E/CameraLogger: Start: about to call onStart() STATE_STARTING
08-05 19:34:51.915 16946-16946/com.some.packed E/CameraLogger: Start: posting runnable. State: STATE_STARTING
08-05 19:34:51.915 16946-16946/com.some.packed E/CameraLogger: Start: posting runnable. State: STATE_STARTING
08-05 19:34:52.034 16946-16946/com.some.packed E/CameraLogger: preview: isHardwareAccelerated: true
08-05 19:34:52.034 16946-16946/com.some.packed E/CameraLogger: preview: isHardwareAccelerated: true
08-05 19:34:52.054 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.054 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.054 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.054 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.057 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.057 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.058 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.058 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.077 16946-16946/com.some.packed E/CameraLogger: onSurfaceAvailable: w= 720 h= 1280
08-05 19:34:52.077 16946-16946/com.some.packed E/CameraLogger: onSurfaceAvailable: w= 720 h= 1280
08-05 19:34:52.079 16946-16946/com.some.packed E/CameraLogger: onSurfaceAvailable: Size is 720x1280
08-05 19:34:52.079 16946-16946/com.some.packed E/CameraLogger: onSurfaceAvailable: Size is 720x1280
08-05 19:34:52.129 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.129 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.129 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.129 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.132 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.132 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.133 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.
08-05 19:34:52.133 16946-16946/com.some.packed E/CameraLogger: onMeasure: surface is not ready. Calling default behavior.

natario1, 2018-08-06:
Hi, please post logs from the blank demo app but only those coming from the demo app package.

Logs should start the moment you open the app

KORuL, 2018-08-07:
it is amazing! I rebooted the phone and now everything works, I can not achieve that effect. Sorry

natario1, 2018-08-07:
Cool. Sometimes when an app messes up with the camera, it breaks the camera support for all other apps.

closed 2018-08-07 by natario1