natario1/CameraView #255: Capture Picture not working on Galaxy S2 API 17

Nucifera8472, 2018-08-08:
- [x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:
1. Install demo app on Galaxy S2 API 17
2. Capture picture

Unfortunately it is our only API 17 device, so I can not check with similar devices. I have not encountered this problem with newer APIs / devices.

#### Expected behavior:
Captured picture is shown

#### Actual behavior:
The preview freezes as soon as you hit the capture button, nothing else happens

#### Interesting logs:
The camera logs stop right after the call of capturePicture. There are a couple of exynos_camera entries afterwards, but nothing else

08-08 16:23:18.300 4342-4342/com.otaliastudios.cameraview.demo V/Camera1: capturePicture: scheduling
08-08 16:23:18.300 4342-4357/com.otaliastudios.cameraview.demo V/Camera1: capturePicture: performing. false
08-08 16:23:18.300 1842-1842/? D/exynos_camera: exynos_camera_get_parameters(0x400682c0)
    exynos_camera_put_parameters(0x400682c0)
08-08 16:23:18.300 1842-2824/? D/exynos_camera: exynos_camera_set_parameters(0x400682c0, effect=none;effect-values=none,mono,negative,sepia,aqua;exposure-compensation=0;exposure-compensation-step=0.5;flash-mode=off;flash-mode-values=off,auto,on,torch;focal-length=4.03;focus-areas=(0,0,0,0,0);focus-distances=0.15,1.20,Infinity;focus-mode=infinity;focus-mode-values=auto,infinity,macro,fixed,facedetect,continuous-video;horizontal-view-angle=60.5;iso=auto;iso-values=auto,ISO50,ISO100,ISO200,ISO400,ISO800;jpeg-quality=90;jpeg-thumbnail-height=240;jpeg-thumbnail-quality=100;jpeg-thumbnail-size-values=320x240,400x240,0x0;jpeg-thumbnail-width=320;max-exposure-compensation=4;max-num-focus-areas=1;max-zoom=30;min-exposure-compensation=-4;picture-format=jpeg;picture-format-values=jpeg;picture-size=3264x2448;picture-size-values=3264x2448,3264x1968,2048x1536,2048x1232,800x480,640x480;preferred-preview-size-for-video=640x480;preview-format=yuv420sp;preview-format-values=yuv420sp,yuv420p,rgb565;preview-fps-range=7000,30000;preview-fps-range-values=(7000,30000);preview-f
08-08 16:23:18.305 1842-2824/? D/exynos_camera: exynos_camera_params_apply: Preview size: 640x480, picture size: 3264x2448, recording size: 720x480
08-08 16:23:18.305 1842-2158/? D/exynos_camera: exynos_camera_enable_msg_type(0x400682c0, 258)

Nucifera8472, 2018-08-08:
captureSnapshot() works, btw

natario1, 2018-09-20:
I have no idea! If you can debug this yourself and see what's wrong, that would be great, otherwise there's nothing we can do here

Nucifera8472, 2018-09-21:
Actually, I tried to debug it just recently, and the thing is, the crash does not happen while in debug mode. It just takes tremendously long to capture the picture. Maybe there is a relevant difference between having the debugger attached and the released version that explains how that can happen.