natario1/CameraView #259: MediaRecorder: stop failed: -1007

tyl210110, 2018-08-14:
This problem often occurs：MediaRecorder: stop failed: -1007

suarezu, 2018-08-21:
yes, this is a fabulous library, but when recording video at the first time or switching from video to picture or from picture to video, it always reports this error

MediaRecorder: stop failed: -1007

and all subsequent same type operations seem successful, please help

wasim5390, 2018-11-22:
Start Recording like this,

natario1, 2018-12-16:
I'm sorry, MediaRecorder fails for a number of reasons and gives no clue about what's wrong, most of the times it's something device specific that we have no control about.

If you can post a demo app that immediately reproduces this issue, I will take a look and reopen.

closed 2018-12-16 by natario1