natario1/CameraView #269: Flash not working

dip1410, 2018-09-17:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ ] *I am using the latest version*
- [ ] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1.Set Flash ON
2.Take a picture
3.Open the picture

#### Expected behavior: Picture with flash

#### Actual behavior: Flash just closes before picture is taken. Picture is always in dark

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

MoustafaElsaghier, 2018-09-18:
Can you please post your code? It works well for me?

natario1, 2018-09-20:
I have seen this happen, but there is nothing we can do. We tell Camera1 to use the flash. Then it sometimes / on some devices will not respect this...
But it's out of our control.

rupesh0895, 2018-10-24:
package com.technetty.myapplication;

import android.annotation.SuppressLint;
import android.content.Intent;
import android.graphics.Bitmap;
import android.graphics.BitmapFactory;
import android.os.Environment;
import android.support.v7.app.AppCompatActivity;
import android.os.Bundle;
import android.view.View;
import android.widget.Toast;

import com.otaliastudios.cameraview.CameraListener;
import com.otaliastudios.cameraview.CameraUtils;
import com.otaliastudios.cameraview.CameraView;
import com.otaliastudios.cameraview.Flash;

import java.io.File;
import java.io.FileNotFoundException;
import java.io.FileOutputStream;

public class MainActivity extends AppCompatActivity {
    CameraView camera;
    private Bitmap bitmap;
    private File file2;
    private String path2;

    @Override
    protected void onCreate(Bundle savedInstanceState) {
        super.onCreate(savedInstanceState);
        setContentView(R.layout.activity_main);
        camera = findViewById(R.id.camera);
        camera.addCameraListener(new CameraListener() {
            @SuppressLint("WrongThread")
            @Override
            public void onPictureTaken(byte[] picture) {
                // Create a bitmap or a file...
                // CameraUtils will read EXIF orientation for you, in a worker thread.
                Bitmap b1 = BitmapFactory.decodeByteArray(picture , 0, picture.length);
              // Bitmap b1= CameraUtils.decodeBitmap(picture, 1000, 1000);
               viewTest.setimage(b1);

               Intent intent=new Intent(MainActivity.this,viewTest.class);
               startActivity(intent);

            }
        });

    }
    public void  clickPic(View view){
     Toast.makeText(this, "capturing Pic",Toast.LENGTH_SHORT ).show();

     camera.setFlash(Flash.ON);
     camera.capturePicture();

    }

    @Override
    protected void onResume() {
        super.onResume();
        camera.start();
    }

    @Override
    protected void onPause() {
        super.onPause();
        camera.stop();
    }

    @Override
    protected void onDestroy() {
        super.onDestroy();
        camera.destroy();
    }
}

closed 2018-12-16 by natario1

pawnjester, 2021-07-26:
@natario1  tried this with the latest version, flash does not come on when video is recording