natario1/CameraView #272: focusMarker slide off screen like a ghost

suarezu, 2018-09-21:
*If you are trying to report a bug, please file the information below.
Prerequisites:*

- [ x] *I am using the latest version*
- [x] *I can reproduce the bug in the demo app, and if not...*
- [ ] *I have tried to reproduce the bug in the demo app*

#### Detailed steps to reproduce the issue:

1. in an androidx fragment's layout add a camera view, no specific attributes set, especially the gesture map:
`
    <com.otaliastudios.cameraview.CameraView
        android:id="@+id/camera"
        android:layout_width="match_parent"
        android:layout_height="match_parent"
        app:cameraJpegQuality="100"
        app:cameraPictureSizeMinHeight="1920"
        app:cameraPictureSizeMinWidth="1080"
        app:cameraSessionType="picture" />`
2. the fragment has a slide in/out transition in its parent activity
3. when the back pressed and the fragment slides out, **the focusMarker shows and slides off like a ghost**

#### Expected behavior:
the focus marker should stay GONE if not mapped whenever it's focusing or not
#### Actual behavior:

#### Screenshots:

#### Interesting logs:

*You can use `CameraLogger.setLogLevel(LEVEL_VERBOSE)`
to add logs to logcat, and `CameraLogger.registerLogger()`
to handle logs yourself (e.g. export to file or a crash reporting service)
so you can collect data from your users.*

natario1, 2018-09-21:
It will just follow your fragment wherever it goes. I think it is mapped by default, try mapping to `none` to disable

natario1, 2018-09-21:
If you can, post a video here for us to see

suarezu, 2018-09-25:
sorry for the holiday off. I just turn TAP off but still the focusMarker shows when sliding. removing 2nd child of camera view doesnt work, now I have to reflect the mTapGestureLayout field to set it GONE.

The video is attached
[image]

natario1, 2018-09-25:
You can set CameraView to invisible before animating. Anyway I will take a look at this!

suarezu, 2018-09-25:
thank you very much for investigating this little fault. wait for the update and use reflection for now; it works.

natario1, 2019-08-21:
I'm closing this as we have a new way to draw markers and this edge case might be fixed now.
If you want to reopen please attach a GitHub repo where the issue is reproducible.

closed 2019-08-21 by natario1