natario1/CameraView #273: StartPreview failed

ritesh94, 2018-09-21:
CameraView Version: 1.6.0

```
Fatal Exception: java.lang.RuntimeException: startPreview failed
       at android.hardware.Camera.startPreview(Camera.java)
       at com.otaliastudios.cameraview.Camera1.applySizesAndStartPreview(Camera1.java:160)
       at com.otaliastudios.cameraview.Camera1.bindToSurface(Camera1.java:134)
       at com.otaliastudios.cameraview.Camera1.onStart(Camera1.java:191)
       at com.otaliastudios.cameraview.CameraController$3.run(CameraController.java:180)
       at android.os.Handler.handleCallback(Handler.java:746)
       at android.os.Handler.dispatchMessage(Handler.java:95)
       at android.os.Looper.loop(Looper.java:148)
       at android.os.HandlerThread.run(HandlerThread.java:61)
```

natario1, 2018-09-21:
Is this the demo app?
Does the app crash?
Are you getting an exception in onCameraError?
Do you have CameraLogger logs?
Please read the issue suggestions.

closed 2018-10-09 by natario1

NoceloEder, 2018-10-16:
I have same issue. I have version 1.6.0 installed.

```CameraLogger``` says this:

```
10-16 16:17:39.593 20748-20800/com.checker.check D/CAM_LOG: Camera1
    onStart: Failed to connect. Maybe in use by another app?
10-16 16:17:39.594 20748-20800/com.checker.check D/CAM_LOG: CameraController
    uncaughtException: Interrupting thread with state: STATE_STARTING due to CameraException: com.otaliastudios.cameraview.CameraException: java.lang.RuntimeException: Fail to connect to camera service
```

Steps to reproduce:
1. Init CameraView ```cameraView = (CameraView) findViewById(R.id.camera) ;```
2. Add ```cameraView.setLifecycleOwner(this);```
3. Add ```CameraLogger```
4. Add ```CameraListener()```
5. Init with FRONTAL CAMERA ```cameraView.setFacing(Facing.FRONT);```(before of this evaluate if device contains frontal camera, otherwise init by default camera "back" ).
6. ```cameraView.start();```

Actually im using CameraView in Activity. All steps are in ```onResume()```

Im test on Emulator. For Android 4.4, 6.0, 7.1, 8.0 this code works fine. **But in Android 5.1, 5.0 crash.**

NoceloEder, 2018-10-16:
When init by default Facing (back camera) and I try to flip to front camera also crash. (same error)