natario1/CameraView #279: Capturing photo/video on Samsung Galaxy Note 8 not working at all

arrabidas92, 2018-10-02:
Prerequisites:

*I am using the latest version*

#### Detailed steps to reproduce the issue:

Try to capture a snapshot/video on Samsung Galaxy Note 8

#### Expected behavior:

Display picture or video preview

#### Actual behavior:

Picture taken or video not displayed and loader is infinite.

The same code run perfectly on other phones like Motorola, Doogee but this issue is only on Samsung devices.
[image]

I also have this issue using directly camera 2 api. If you have some advice, I take it.

Namazed, 2018-10-08:
Hi, @arrabidas92.
Did you used CameraLogger? Can you show your logs when you taking a picture or video?

arrabidas92, 2018-10-09:
Finally I solved my issue by disabling hardware acceleration.
Now it works !

closed 2018-10-09 by arrabidas92