natario1/CameraView #280: Exif information is null

XuJian1252878, 2018-10-03:
First,Thanks for you to provide the CameraView Library.
I am using the latest version, and I found that I can't read the image's exif information after I save the image through CameraView (This situation doesn't exist in the System camera, I have tested...).

my code is as following:

`
CameraUtils.decodeBitmap(jpeg, nativeWidth, nativeHeight, new CameraUtils.BitmapCallback() {
            @Override
            public void onBitmapReady(Bitmap bitmap) {
                // save the token photo
                FileUtils.saveImgBitmap(resImgPath, bitmap);
                CameraOperator.builder()
                        .setCameraResImgPath(resImgPath)
                        .start(CameraFragment.this.getActivity());
            }
        });
`

about the save image method:
`
public static boolean saveImgBitmap(String path, Bitmap bitmap) {
        File file = new File(path);
        try {
            if (!file.createNewFile()) {
                return false;
            }
            FileOutputStream fos = new FileOutputStream(file);
            bitmap.compress(Bitmap.CompressFormat.JPEG, 100, fos);
            fos.flush();
            fos.close();
            return true;
        } catch (IOException e) {
            e.printStackTrace();
        }
        return false;
    }
`

then I get the exif information:
`
// for example: date
ExifInterface exifInterface = new ExifInterface(imagePath);  // exifInterface is not null
String dateStr = exifInterface.getAttribute(ExifInterface.TAG_DATETIME);  // dateStr is null
`

the `dateStr` is null, the return values for `TAG_GPS_ALTITUDE, TAG_GPS_LATITUDE ...`  are also null. I don't why.

natario1, 2018-10-03:
Bitmaps do not include metadata, so if you use Bitmap you will lose it. You can save to file the byte[] array directly

XuJian1252878, 2018-10-03:
> Bitmaps do not include metadata, so if you use Bitmap you will lose it. You can save to file the byte[] array directly

Thanks

closed 2018-10-03 by XuJian1252878

Nek-12, 2021-09-01:
This should be mentioned in the docs @natario1 
Spent a lot of time figuring it out

dany171, 2024-02-27:
Is it possible to get Exif data from the bytes?
Asking because I am taking the bytes from CameraListener.onPictureTaken(PictureResult.data) and using android.media.ExifInterface(pictureResult.data.inputStream()) to get Exif data, however it only comes with ImageWidth, ImageLength, Orientation and LightSource, that is it, I want to get the full Exit attributes values.